r-lidar/lidR #105: catalog_reshape with buffers

harryjfowen, 2018-02-05:
Hi,

I have some large terrestrial laser scanner point clouds that I wish to split into smaller chunks - the catalog_reshape function is very useful but I wish to split my LAS file into smaller chunks that each include an overlapping buffer and output as new .las files. These will then feed into some denoising routines. Is this possible?

Thanks

Harry

Jean-Romain, 2018-02-05:
Hi,

You can process your catalog as you described using `catalog_apply` but you can't reshape and save las file using a buffer with `catalog_reshape`. This is because I enforced the buffer to 0 for this specific function and I will not change that until I find a very good reason to do it.

However if you really want it you can download the source, tweak the code and make your own version of the code to enable such feature easily. In `catalog_reshape.r` line 63, change the `0` to the expected buffer size. For example:

```r
 clusters = catalog_makecluster(ctg, 1, 25, FALSE, size)
```

And it should work.

harryjfowen, 2018-02-05:
Great! thanks for the help

closed 2018-02-05 by Jean-Romain