r-lidar/lidR #106: warning's points below zero after normalization lesser than the actual number in the dataset

bangwim, 2018-02-16:
Hi Jean,

I am contacting you again to clarify more about the warning raised after normalization process which tells us about some points below 0 are exist.  Previously you explained very nicely with illustration how rasterization I am sorry that I cannot give you a reproducible example due to a large number of points.  But I put the code as follows with the hope that you could follow what I meant still:

`las2 <- readLAS("AREA_2_ALL_CLASS.las", filter = "-drop_z_below 0")`
`dtm2 <- grid_terrain(las2, res=0.5, method="kriging", k=15, keep_lowest = T)`

I checked 'dtm2' and normalized 'las2' objects (copy=T)  whether they have 'Z' value below zero.

`dtmdt <- as.data.table(dtm2)`
`dtmdt[dtmdt$Z < 0]`
`# Empty data.table (0 rows) of 3 cols: X,Y,Z`

So, no zero found on the DTM of which I think that "-drop_z_below 0" filter worked.

`# by discretization`
` dctlasnorm <- lasnormalize(las2, dtm2, copy=T)`
`# Warning message:`
`# 2678 points below 0 found.`

`# by point clouds interpolation`
`itplasnorm <- lasnormalize(las2, method="kriging", k=15L, copy=T)`
`# Warning message:`
`# 2677 points below 0 found.`

Both normalization methods showed warnings.  But why they differ with the result of the number of points below zero through filtering the normalized 'las2',  either by discretization and interpolation, as shown below:

`nrow(dplyr::filter(itplasnorm@data, Z < 0))`
`# [1] 185068`

`nrow(filter(dctlasnorm@data, Z < 0))`
`# [1] 189037`

My further questions:
(1) Which one is right? The previous warnings OR the number of points below zero that filtered directly from normalized las datasets?
(2) Would you suggest me to filter out all of those points below zero?  It's a quite significant number I think.  
(3) We did not create a checkpoint on the ground to calculate the precision of elevation measurement.  Would it be sensible to justify 1%-5% threshold (as zero) as it's not easily be distinguished on the ground and trim all the remaining smaller values?  But it's a lot still:

`lt0 <- filter(itplasnorm@data, Z < 0)`
`nrow(filter(lt0, Z < 0))`
`# [1] 185068`
`nrow(filter(lt0, Z < 0 & Z > -1))`
`# [1] 182391`
`nrow(filter(lt0, Z < -0.05))`
`# [1] 139200`

[image]

`summary(lt0$Z)`
`#    Min. 1st Qu.  Median    Mean 3rd Qu.    Max. `
`# -7.9200 -0.2390 -0.1160 -0.1906 -0.0460 -0.0010 `

`lt0las <- lasfilter(itplasnorm, Z < 0)`
`# Warning message:`
`# 2677 points below 0 found. `
`plot(lt0las)`

[image]

I am thinking to send the whole dataset that I used here for you to check if approved by our research leader.  I would appreciate much any comment from you.

Best regards,
Wim
(Forests2020 Indonesia)

Jean-Romain, 2018-02-16:
Hi bangwim,

Many question. Hopefully they are not actual issues. I will try to answer to your question the best I can.

> I checked `dtm2` and normalized `las2` objects `(copy=T)` whether they have Z value below zero.

Why  `dtm2`  would have negative values? This is an image of the topography. The average value is maybe 500 m hight (or more or less depending of the elevation of your region). Maybe few points are negative (elevation below the sea level) but this is not the most common.

> So, no zero found on the DTM of which I think that `-drop_z_below 0` filter worked.

Sure it works. But it simply do not read points with negative value in the las file. Considering that your file is not normalized you probably don't have negative points.

> Both normalization methods showed warnings. But why they differ with the result of the number of points below zero through filtering the normalized las2, either by discretization and interpolation, as shown below:

Yes because the methods are different.  As you said I already show you a picture that explain the main principle.

> Which one is right? The previous warnings OR the number of points below zero that filtered directly from normalized las datasets?

That is a good question. Your result is weird.

> Would you suggest me to filter out all of those points below zero? It's a quite significant number I think.

I do not suggest anything. I only explain how algorithms work if it not clear enough for the user. I don't have any authority to tell you what to do. Considering that the scientific literature is usually very weak on the materials and methods I can't tell you what are the "usual" practices.

So, as a conclusion it remains 1 question. Why the number of negative points given by the warning message do not match the number of row of your filter table after your filter command? 2700 negative point sound quiete normal. But your plot clearly shows that 18000 is more likely to be the real number. It seems to concern the whole dataset. Maybe you can share only a subset of your dataset in laz format to help me to reproduce the bug.

Jean-Romain, 2018-02-16:
I think I got it. I introduced a bug in the count of points below 0 by speeding up the code. I will update in few minute and you will tell me if your results are better

bangwim, 2018-02-16:
Dear JR, 

Thanks a lot for your quick response.  Sorry for those questions.  
Anyway, I reinstalled lidR through github.

And here is I got:

`dctlasnorm <- lasnormalize(las2, dtm2, copy=T)`
`# Warning message:`
`# 189037 points below 0 found.

`nrow(filter(dctlasnorm@data, Z < 0))`
`# [1] 189037`

So, now the warning gives the same number of points with the number from filtered table command number.  It means that it might be something wrong with the data.
I got permission from my principal to share the data.  I will contact you again when it's ready.
Thanks a lot for your kind support.

Best regards,
Wim

Jean-Romain, 2018-02-16:
19000 point below 0 is a lot I do agree but regarding the original dataset it is only 0.3%

Jean-Romain, 2018-02-16:
If you share the data I will have a look but I'm confident that everything is ok.  This is just normal and expected considering the inaccuracies of the ground classification and the inaccuracies of the interpolation itself. Just for a hint could show me the classification of the negative points.

```r
neg = lasfilter(dctlasnorm, Z < 0)
head(neg@data)
plot(neg, color = "Classification")
```

Also you can try to normalize with another software and tell me if the number of negative values is lower.

closed 2018-02-16 by Jean-Romain