r-lidar/lidR #107: Exporting the segmented crown into shapefile: Li2012

karnayogendra, 2018-02-20:
Hi Jean,

I am using several methods of lidR package to segment the tree crown and export it into the shapefile. I have successfully run the `lastrees_watershed,` `lastrees_dalponte` and `lastrees_silva` and could export the output crown as a shapefile but couldn't export the output from `lastrees_li` as a shapefile.  Here are the codes, results and error that I obtained.

`liseg <- lastrees_li(Rnorm, dt1 = 1.5, dt2 = 2, hmin = 2, R = 10)`
`liseg`
NULL
`crown.shp <- rasterToPolygons(liseg, fun=NULL, n=4, na.rm=TRUE, digits=12, dissolve = TRUE)`

> Error in (function (classes, fdef, mtable)  :   unable to find an inherited method for function ‘nlayers’ for signature ‘"NULL"’

So, I would like to know how can I export the tree crown as a shapefile so that I will compare different segmentation results. 

Thanks and Regards,
Yogendra

Jean-Romain, 2018-02-20:
The li et al algorithm is a point cloud based method. The point cloud is segmented and updated. Each point get an ID that refer to a single tree and the function returns nothing because there is nothing to return. This is documented in the package. So your question do not have any anwser because there is nothing to write in a shapefile but a point cloud. Shapefiles can store point clouds so maybe you can use the function `as.spatial` to convert the `LAS` object into a `SpatialPointDataFrame` and then write it into a shapefile. But Im pretty sure this is not very useful.

closed 2018-02-20 by Jean-Romain

slpgnt, 2018-03-13:
Here is a clunky solution that lacks elegance. If you re-attribute the intensity field with the lastrees created treeID field then write a new .las file you can then run it though the lasthin.exe and las2dem.exe (LASTools using the Intensity field) in arc map to generate a raster. The raster can then be converted into a polygon and smoothed in arcmap. 

Alternately you could write an LAS to Raster function in R, or load the re-attributed LAS file into Arc as a LASDataset and run the las to raster function.

Jean-Romain, 2018-03-13:
@slpgnt in that case you can use `grid_metrics` to rasterize within R. Also within R too, you can compute a convex hull for each tree. There are many post processing options that are free to the user.

Also extra bytes attributes are almost supported in `rlas` (development version). Thus writting `treeID` will become possible soon.

slpgnt, 2018-03-13:
That’s great news!