r-lidar/lidR #108: error with lastrees

ervanSTRI, 2018-02-20:
Dear Jean-Romain,
First of all thanks for this impressive package. Super efficient and clean! Love it. I am using some of its functions to delineate tree crowns in Switzerland, mainly on deciduous species (Quercus). As crown are wider than Douglas fire, finding treetops is more challenging. I got my way through this, but once I want to delineate the crown perimeter I got the following error message:

lastrees(las, "watershed", chm, th = 4,extra=T) # segmentation
Error: 'EBImage' package is needed for this function to work. Please read documentation.
> install.packages('EBImage' )
Installing package into ‘C:/Users/Ervan/Documents/Work/R/R.library’
(as ‘lib’ is unspecified)
Warning in install.packages :
  package ‘EBImage’ is not available (for R version 3.4.1)

Do I need to downgrade R, or am I doing anything wrong?? 
Merci pour votre retour et tout le temps passé!
Amitiés,
Ervan

Jean-Romain, 2018-02-20:
Hi Ervan,

This question is not related to the `lidR` package but to the `EBImage` package. About the watershed method the `lidR` documentation says:

> This method is a simple watershed segmentation method. It is based on the bioconductor package EBIimage. You need to install this package to run this method (see its [github page](https://github.com/aoles/EBImage)).

And the `EBIimage` package explain how to install this package:

```r
source("https://bioconductor.org/biocLite.R")
biocLite("EBImage")
```

Indeed this not a regular CRAN package but a bioconductor package. Thus `install.package` tell you that `EBImage` does not exist.

closed 2018-02-20 by Jean-Romain

ervanSTRI, 2018-02-21:
Hi Romain,
Thanks for pointing towards the package repository. Now that everything is installed, I got the following error message:
trees <- lastrees(las, "watershed", chm, th = 5,extra=F) # segmentation
Error in (function (classes, fdef, mtable)  : 
  unable to find an inherited method for function ‘extent’ for signature ‘"lasmetrics"’

Any hint? Thanks
E

Jean-Romain, 2018-02-21:
I guess I missed something in the code. Try:

```r
trees <- lastrees(las, "watershed", as.raster(chm), th = 5,extra=F)
```

ervanSTRI, 2018-02-21:
Sorry, I missed that line. My fault.

ervanSTRI, 2018-02-21:
Ah! I noticed a typo here:
crown = lastrees(lasnorm, "watershed", chm, th = 4, extra = TRUE)

library(raster)

# get contours as a SpatialPolygonsDataFrame
# (take a while to run)
contour = rasterToPolygons(crown**s**, dissolve = TRUE)  ## should be "crown" not "crowns"

Jean-Romain, 2018-02-21:
where ?

ervanSTRI, 2018-02-21:
https://github.com/Jean-Romain/lidR/wiki/Tree-segmentation-from-A-to-Z
The "Optionally" section

Jean-Romain, 2018-02-21:
Ok thank for reporting. There are probably many errors in the wiki pages. I cannot keep everything up-to-date alone (package + doc + wiki) :disappointed: .  Do not hesitate to modify the wiki yourself without my permission if you find an error, typo or anything else :wink: