r-lidar/lidR #109: Repeating TreeID

slpgnt, 2018-02-23:
Hi JR

As you are likely aware, when catalog apply is run (by tile or by file) the iterations repeat the tree ID per tile or file. Is there a way around this? What I am looking for is unique tree ID's for each tree across an entire catalog be it repeating tree ID with a tile modifier or a complete unique ID. 

Thanks in advance for your time!

bi0m3trics, 2018-02-23:
I'm sure JR will have a more elegant solution, but an easy fix would be to just renumber the tree ids after you `rbindlist `the resulting `data.table` from `catalog_apply`. For example (...from the `catalog_apply` example in the help)
```
output = catalog_apply(project, tree_area, fargs, select = "xyz")
output = data.table::rbindlist(output)
# now replace the tree column with a sequence of (unique) numbers, the tree ids are arbitray at this point anyway!
output$tree = seq.int(nrow(output))
```
If you need them to be unique within the clusters assigned by catalog_apply... that's a different story and I would wager a much harder nut to crack. 😭

slpgnt, 2018-02-23:
Thank you bi0m3trics. That is a great solution but unfortunately wont work for my situation. I should have explained my predicament in more detail. 

What I need is the remuneration of the points to be done within the catalog apply function (or the analysis custom function). What I am trying to do is normalize the tree height (0-1) for each tree. Thus taking the max and min values for say tree 1 would actually be wrong as it is taking from all tree 1's. As such something like a tile modifier could work. I've tried to generate a new column which contains the input files ID but can never get it to work!...

Thanks again for your input!

bi0m3trics, 2018-02-23:
Weird... Now I feel like I'm missing something... I'm pretty sure that the treeIDs assigned by `lastrees` are unique within a cluster, so getting min and max heights (`min(z)` and `max(z)`) would work fine within a cluster (that's how the `myMetrics` example works) and would only appear to repeat numbers once you post process the results.  I'll be quiet now... 😉

Jean-Romain, 2018-02-23:
Hi, 

You pointed out a known limitation that I did not solved yet and which is not trivial. There are two things to consider:

1. `lastrees` returns (actually updates) a point cloud. Thus, applying `lastrees` on a catalog is almost  the same as loading the entire catalog and it requires a substantial amount of memory.
2. attributing a different id to each tree implies to know what was the lastest id found to start at this id on the next file. This is not possible in parallel.

So, what are the solutions?

1. Enable to save a las file that stores of the tree  the IDs. We are working on it in the `rlas` package.
2. Create a unique ID for each tree for example combining the X and Y coordinates of the highest point of each tree in a single big number. This is an elegant solution and I plan to think more about it.

So it might become possible. But at the time being I don't have a plug and play solution for you. However you can, at least, implement the solution 2 relatively easily.

Regards

slpgnt, 2018-02-23:
Ha!, Bi0m3trics and JR thank you for your help. I may have over though this issue a bit! 

Have a good weekend!

Jean-Romain, 2018-02-23:
@bi0m3trics your solution works only if you return a single number per tree (area for example). Thus a tree id is just a row. @slpgnt  was asking for relabelling the point cloud.

closed 2018-02-23 by Jean-Romain

slpgnt, 2018-03-12:
Hi Jean,

In your Li2012 Algorithm are the dt1 and dt2 parameters the same as the dmin1 and dmin2 parameters in Li et.al. 2012 paper where you found the algorithm?

Thank you

Closed #109<https://github.com/Jean-Romain/lidR/issues/109>.

Jean-Romain, 2018-03-12:
In Li's paper dmin1 et dmin 2 are not parameters. `dt1` and `dt2`correspond to the parameter ![](https://latex.codecogs.com/gif.latex?d_t) that may change depending on the point elevation.

slpgnt, 2018-03-12:
Okay, as I expected.  Does your code observe the same Z limits (15m) as proposed in his paper?

where dt is an adaptive threshold:
dt = 2 m if Zu !15 m;
dt = 1.5 m if Zu "=15 m

Jean-Romain, 2018-03-13:
Yes it does strictly what is written in the paper. But I'm thinking about adding a third parameter to allows the user to change the 15m threshold.

slpgnt, 2018-03-13:
That would be of interest.

Jean-Romain, 2018-03-13:
It can be done within 30 minutes if you want it

slpgnt, 2018-03-13:
Okay. However, I am not in strict need of it at this time but would be interesting to see how changing this value affects the segmentation.

Jean-Romain, 2018-03-13:
Done fa22fae1

slpgnt, 2018-03-13:
Thank you

Done fa22fae<https://github.com/Jean-Romain/lidR/commit/fa22fae196a2752bd58b45539f5eff8c6746a069>