r-lidar/lidR #111: Transform LAS coordinates

spono, 2018-03-01:
Hi JR,
any suggestion on how to properly handle the CRS transformation of a las file?

I tried the following script but its results pretty nasty, mostly because the header is not updated according to new coordinates. 

`las = readLAS(file)`
`hd = las@header # header`
`las = as.spatial(las)`
`proj4string(las) <- CRS(projectCRS)`
`las = spTransform(las, new_projectCRS) # convert from WGS84 to UTM32`
`las = cbind(las@coords, las@data)`
`las = LAS(data=las, header=hd)`
`las@crs <- sp::CRS(new_projectCRS)`

Jean-Romain, 2018-03-01:
What is the problem? Just reading the code, I'm expecting to get a correct output.

spono, 2018-03-01:
coordinates are indeed correct, but if I try to plot the `catalog` of the transformed files it can't be displayed over the usual google map. Plotting the single `las` it is scaled incredibly small and incorrect.

So I looked at the header and the only strange thing is that `offset `seems to remain the same:
`Scale factor X Y Z:       1e-07 1e-07 0.001` 
`Offset X Y Z:             11.62887 45.25966 0`

I personally don't know where else to look for issue.

Jean-Romain, 2018-03-01:
A catalog is not related directly to the data you loaded into R. A catalog in only files headers. Sometime the CRS in not properly found so you can change it to set the good one. Thus the catalog will be display at the good place. You can try to manually re-project the catalog but it cannot affects the files.

If you re-project manually a LAS object as you did, there no way to change anything in the header but `Min X` `Min Y` `Max X` `Max Y`. And the header does no affect how the file is displayed. So, I don't understand what do you mean by "scaled incredibly small".

So I don't really understand the issue.

Jean-Romain, 2018-03-01:
I think I got it. You converted a metric CRS into a non-metric (degree?) CRS. But you converted only X and Y but not Z. Thus you have X Y in degrees and Z in meters and the point cloud is broken.

Jean-Romain, 2018-03-19:
Did we solve your problem by email? It was a question of degree converted in meters right? Can I close the issue?

spono, 2018-03-20:
Oh gosh, yes, sorry: I thought i replied to your email!
Yes, it works but still some tiles have some issues, but (i think) related to other problems.
Thank you again

A: Jean-Romain/lidR
Cc: Nic; Author

closed 2018-03-20 by Jean-Romain

zoeschindler, 2021-06-08:
Hello, I am currently having the same issues. I am trying to transform my EPSG:4978 point cloud into a EPSG:25832 point cloud, but when I try to do so, I get broken point clouds. I tried saving the z-attribute as additional attribute and reassign it after the transformation, but since Z is a protected attribute, I did not manage to do so. Can you tell me how the problem was resolved?

zoeschindler, 2021-06-08:
I am sorry, it just turned out to be my mistake - at least I think so.