r-lidar/lidR #112: Retain source file name within catalog_apply

brycefrank, 2018-03-01:
Hello,

I am trying to retain trees after using tree_detection that are only within the 0th buffer. I think the most straightforward way, and a nice feature, would be to retain the path of the source las file (i.e. the path of the 0th buffer source) in the LAS object that gets passed upon each iteration of catalog_apply that could get accessed in a function.

My understanding is that only a LAS object gets passed for each iteration of catalog_apply, and adding some additional information to that object should be marginal from a computational stand point.

Something like:

    las@source_path

Jean-Romain, 2018-03-01:
Hi, thank your for this suggestion. Your point is correct and  this is a known issue. I'm trying to figure out a good solution. 

Your proposition, sadly, is not workable because a `source_path` attribute do no necessarily have a meaning. Indeed the las object you get can have been read from several different files. Thus `source_path` is meaningless in some case and cannot be used as a solution.

A solution would be to generate a random name. That could technically works but in my opinion is a poor idea.

So far I know that I must work on this feature but I cannot find an good solution without breaking compatibility (actually I don't have a solution at all). Feel free to suggest something.

brycefrank, 2018-03-01:
If my understanding is correct, the 0th buffer is just the original LAS object before buffering. Is this information not retrievable before the buffering process begins?

Jean-Romain, 2018-03-01:
No, it is a subregion of the whole catalog. You chose the size of the subregion as a function of your available memory.

The size of this subregions is defined in the global options (`catalog_option`). Because the subregion all together create a continuous set, a subregion often fall between two or more files. The only case for which your assumption is true is when the catalog is processed by file (`by_file = TRUE`).  But this is a specific case.

brycefrank, 2018-03-01:
I see, I see.

At some point in the next few months I will need to process with this concept in mind. If I find a solution or at least a good workaround you will be the first to know.

Jean-Romain, 2018-03-02:
The same for me. I will keep this issue opened and add information if something changes.

Jean-Romain, 2018-03-19:
I got an easy to implement idea. When you load LAS object in `lidR` it is expected to be a strictly valid representation of the a `las` file. Header is generally updated when the original file is corrupted or wrongly built.

A valid las file has a [guid](https://fr.wikipedia.org/wiki/Globally_Unique_Identifier). For now `rlas`does not read and write properly this information thus it is no usable. But the next release will support properly the `guid`. Thus the `guid` can be use as a unique identifier of any `LAS` object even the cluster load sequentially in `catalog_apply`.

Jean-Romain, 2018-08-15:
Well some new about the feature. This will be entirely supported in `lidR v2.0` in a very flexible way. Something like that:

This applies the user function on the catalog, processing cluster of 500 x 500  and returns the output in R

```r
ctg = catalog(...)
tiling_size(ctg) <- 500
buffer(ctg) <- 0
catalog_apply(ctg, user_function)
```

This applies the user function on the catalog, processing cluster of 500 x 500  and writes the output in files named after the bounding box of the cluster

```r
ctg = catalog(...)
tiling_size(ctg) <- 500
buffer(ctg) <- 0
output_files(ctg) <- "/document/analsye/cluster_{XLEFT}_{YBOTTOM}_metrics_zmean"
catalog_apply(ctg, user_function)
```

This applies the user function on the catalog, processing the datset by respecting the original file and writes the output in files named after the original las file

```r
ctg = catalog(...)
by_file(ctg) <- TRUE
buffer(ctg) <- 0
output_files(ctg) <- "/document/analsye/{ORIGINALFILENAME}_normalised"
catalog_apply(ctg, user_function)
```

You got the idea. And this will be available for all the functions of the package

brycefrank, 2018-08-15:
Awesome, that should do the trick.
Thanks, JR!

Jean-Romain, 2018-08-17:
Actually it does more than *the trick*. For example if you have a shapefile with polygons describing some stands of forest and the table of attribute have a field  `STANDNAME` you can do:

```r
spdf = readOGR("some/shapefile")
ctg = catalog(...)
cores(ctg) <- 4L
output_files(ctg) <- "/documents/clip/project_stand_{STANDNAME}"
lasclip(ctg, spdf)
```

To clip all you polygons and write them in files named :

- `"/documents/clip/project_stand_name1.las"`
- `"/documents/clip/project_stand_name2.las"`
- `"/documents/clip/project_stand_name3.las"`

If your shapefile contains a field `FORESTTYPE`

```r
output_files(ctg) <- "/documents/clip/project_stand_{STANDNAME}_{FORESTYPE}"`
```

To clip all you polygons and write them in files named :
- `"/documents/clip/project_stand_name1_conifer.las"`
- `"/documents/clip/project_stand_name2_conifier.las"`
- `"/documents/clip/project_stand_name3_deciduous.las"`

closed 2018-08-17 by Jean-Romain