r-lidar/lidR #113: Column skipped when using writeLAS

doublekwok, 2018-03-15:
Hello,

I've used 'lastrees' to get a treeID column, after that, I was trying to export the objects with treeID into a new las file. but I got the warnings below. How can I sort it out?

>writeLAS(.las = tree, file = "modified_test.las")
Column pulseID skipped. It does not meet las specifications.
Column Zref skipped. It does not meet las specifications.
Column treeID skipped. It does not meet las specifications.

Many thanks.

Jean-Romain, 2018-03-15:
Hi,

This is a recurrent question asked by users. Here a clear and definitive answer for you and the next ones :wink: 

A `las` file is a standardized file that have some strict specifications (see [las specifications](https://www.asprs.org/a/society/committees/standards/LAS_1_4_r13.pdf)). A `las` file can contain some data for each point such as X Y and Z coordinates on 32 bits each as well as an intensity value on 16 bits and so on. Everything is rigorously described in the specifications.

So basically a `las` file cannot stores a data that is not in the specs. An ID for each tree is not in the spec and is therefore not writable in a `las` file.

But... the specifications allow for writing extra bytes along each points. Everything is documented, again, in the las specifications. The `rlas` package, that drives I/O of las files, does not support this part of the specs yet. Thus, you can't write `treeID` in a las file. But we are working on it and it is almost done btw. The current development version of `rlas` almost supports extra bytes attribute in writing mode.

Thus, I'm expecting to enable the user to write the `treeID` soon. But this is not currently possible.

Jean-Romain, 2018-03-22:
@doublekwok @karnayogendra  @spono

The current `devel` branch supports for writting `treeID` natively

```r
LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las = readLAS(LASfile, select = "xyz", filter = "-drop_z_below 0")

lastrees(las, "li2012", R = 5)

f = paste0(tempfile(), ".las")
writeLAS(las, f)         

las2 = readLAS(f)
```

Update the `rlas` package from its `devel` branch and `lidR`from the `devel` branch as well. They are "in development" versions but are expected to works fine. If you encounter bugs, please report.

closed 2018-03-30 by Jean-Romain