r-lidar/lidR #115: Creating polygons for individual tree detection

mxblsdl, 2018-03-22:
I have been using the `lastrees` function to plot individual trees with the "li2012" method. You provide a good example of how to use the "watersheds" method to create spatial polygons. 
https://github.com/Jean-Romain/lidR/wiki/Tree-segmentation-from-A-to-Z

When I try to run the "li2012" method it returns a NULL value;
`crowns = lastrees(lasnorm, "li2012", R=5)`
Is there a workflow for creating similar polygon outputs with the other methods in `lastrees`?

bi0m3trics, 2018-03-22:
Here's a modified version of JR's the catalog_apply example that also returns polygons obtained from a call to chull (it's not pretty but it does the job... and allows you to see how you might modify the custom function to get what you actually want).  It then goes on to post-process the data, write the tree location points and polygons to separate .shp files, and plots the results in ggplot. 

It's a clunky solution, but you'll get the point...

[http://quantitativeecology.org/files/r/catalog_apply_exportlocs&polys.R](http://quantitativeecology.org/files/r/catalog_apply_exportlocs&polys.R)

[image]

Jean-Romain, 2018-03-22:
Hi,

This is a recurrent question. See also my answer here #107 . In brief, Li's method is point cloud based method. There is no intermediate object to return and many ways to post-process the output. This is the job of the user to choose how to post-process the output. Li didn't describe this step and thus I didn't not code it.

Also @bi0m3trics's display looks nice. Especially I'm interested to figure out how you computed non convex polygons. I can see in the code a `chull` so I'am really wondering were the non-covex hulls come from. Mom comvex hulls are one of the major difficulties for exporting a shapefile imo.

Jean-Romain, 2018-03-22:
Also the fact the `lastrees`returns `NULL` is documented

> ### Value
> Nothing, the point cloud is updated by reference. If extra = TRUE algorithms provide extra outputs. Usually intermediate objects used internally, such as a RasterLayer.

bi0m3trics, 2018-03-22:
@Jean-Romain I'm not sure I follow as there shouldn't be any non-convex hulls in that output (it could just be overlapping hulls making it "look" non-convex)... As you said, I just used chull (Eddy's quickhull algorithm) so they all should be convex. 

Another option, which I'd advise to use cautiously, would be to explore using alphahulls (sensu [this post](http://r-sig-geo.2731867.n2.nabble.com/alpha-hull-ahull-to-polygon-shapefile-td7342734.html)).  

However, chull works for my purposes so I've yet to take this any further.

closed 2018-03-30 by Jean-Romain