r-lidar/lidR #116: A (maybe) new bug when using catalog_apply()

jequihua, 2018-04-03:
I'm trying to use catalog_apply(). It seems to run fine but on reaching progress 100% I get the following error:

**Error: unable to find an inherited method for function 'area' for signature '"numeric"'**

I even tried the code+data in issue #99 

grid_terrain() works fine with multicore=1 or greater

But catalog_apply() fails for me in both cases.

Running windows 10, R 3.4.3

Jean-Romain, 2018-04-03:
Hi, thanks for reporting an issue. Could you provide a code and a context for your error. I tried the function `area` and it worked fine. I found something incorrect in the package and I must fix it but surprisingly it does not affect the regular behavior of the function `area`. So for now I cannot reproduce the issue without a reproducible example.

jequihua, 2018-04-03:
Yes of course, for example if I run the code provided in issue #99  (on the data also provided):

library(lidR)
lidr_options(progress = TRUE, verbose = TRUE)

project = catalog("./")
project@crs<-sp::CRS("+proj=utm +zone=12 +datum=NAD83")
plot(project)

catalog_options(progress = TRUE, buffer = 10, multicore = 1, tiling_size = 100, by_file=FALSE)

# first with grid_terrain, multicore = 1, works every time, multicore >= 2, has never worked for me as far as I can remember
####################### Works fine ###########################
dtm = grid_terrain(project, res = 1, method = "knnidw", k = 6L)
##########################################################################

plot(dtm)

# and now with catalog_apply, just using the (slightly modified) code from the documentation
tree_area = function(las)
{
  if (is.null(las))
    return(NULL)
  lasnormalize(las, method = "knnidw", k = 8L)

  las %<>% lasfilter (Z > 2)
  if (is.null(las))
    return(NULL)

  lastrees(las, algorithm = "li2012", dt1 = 2, dt2 = 2, hmin = 2, R = 15)
  m = tree_metrics(las, myMetrics(X, Y, Z, buffer))
  m = m[minbuff == 0 & maxbuff <= 2]
  m[, c("minbuff","maxbuff") := NULL]
  return(m)
}

myMetrics <- function(x, y, z, buff)
{
  i = which.max(z)
  xcenter = x[i]
  ycenter = y[i]
  A = area(x,y)
  minbuff = min(buff)
  maxbuff = max(buff)
  
  return(
    list(
      x = xcenter,
      y = ycenter,
      area = A,
      minbuff = minbuff,
      maxbuff = maxbuff
    ))
}

#######################  unable to..."area" for signature "numeric"  error produced #####
output = catalog_apply(project, tree_area, select = "xyzc")
...
Progress: 100%
Error: unable to find an inherited method for function ‘area’ for signature ‘"numeric"’

##########################################################################

If I change in catalog_options to multicore >= 2 the same thing happens.

Jean-Romain, 2018-04-03:
This code is not mine. But it works on my computer. And I guess it works on @bi0m3trics  computer too (otherwise he wouldn't have shared it). Which version of the package are you using ?

jequihua, 2018-04-03:
Version 1.4.1

bi0m3trics, 2018-04-03:
Try changing the area call to lidR::area. I bet you have something loaded that has an area function that doesn't work quite right, and I know JR's works on both of my windows machine and my Linux box.

Jean-Romain, 2018-04-03:
@bi0m3trics That makes sense

jequihua, 2018-04-03:
Yes, it's caused by loading the raster package which has it's own area function. Everything is working fine now. Thanks so much for the replies.

Jean-Romain, 2018-04-03:
I'll try to fix that and make the area function compatible with `raster`

Jean-Romain, 2018-04-03:
This issue has been fixed in `lidR 1.5.0`.

commit a3d51340 (2018-04-03): Inherit area from raster fix #116
closed 2018-04-03 by Jean-Romain