r-lidar/lidR #117: lasclip inside option

rafaelwueest, 2018-04-03:
Hi Jean-Romain

Thanks for your work on lidR, very useful package.
I used `lasclip()` with the `inside = FALSE` option with lidR version 1.3.1 very often and successfully.
It appears that lidR to 1.4.2 does no longer support the `inside` option? I see it still described in your [wiki](https://github.com/Jean-Romain/lidR/wiki/lasclip).

Do I have to live without the `inside = FALSE` functionality or do you have plans to integrate this in some future versions again?
Or can I somehow get to the old  `lasclip(..., inside = FALSE)` functionality via other lidR functions?

Jean-Romain, 2018-04-03:
Hi, somebody asked me the same question few days ago.

`inside` argument has been removed since `lasclip` and other related functions support to clip from a `catalog`. It is still described in the wiki because the wiki is not anymore up-to-date. I don't have time to review the whole wiki pages :-(

Why I did that? Because `lasclip` with `LAS` object works at the R level. And it works with a `LAScatalog` at the C++ level. It was more complex to preserve this behavior for  the two type of queries and, in addition, I did not believe it was useful. I was wrong.

So I will reintegrate this feature as soon as possible.

rafaelwueest, 2018-04-03:
Sounds great, thanks!

Jean-Romain, 2018-04-03:
Well, I remember now why I disabled the option. I **can't** enable it for `LAScatalog` objects. Thus, the function cannot have the same options depending on the input. If the data is loaded into the R memory I can enable  the `inside` option but not if it is a query from a set of file not loaded into the R memory I can't. For consistency I removed the option. 

If I enable it again this will not be available for a `LAScatalog` objects. Only for `LAS`objects. In the same time `inside = FALSE` may returns a huge amount of data for a `catalog`. It might be better to do not provide the option to the user.

rafaelwueest, 2018-04-03:
I see...

> It might be better to do not provide the option to the user.
If this relates to the danger of huge amounts of data; this would only apply for the `catalog` case, right`? No danger for the in-memory case.
If you are concerned about consistency: I would still appreciate the option being there for in-memory use of `lasclip`... Looks like you to agree as it's a To Do now in the Roadmap - would be cool to see it back in!

Jean-Romain, 2018-04-03:
The option is back in version 1.5.0 (`devel` branch).

closed 2018-04-03 by Jean-Romain