r-lidar/lidR #122: problem with lastrees

Indu-cherukuri, 2018-04-17:
> lastrees(lasnorm, "watershed", chm, th = 4,extra=T) 
Error in stopifnotlas(las) : could not find function "stopifnotlas"

Jean-Romain, 2018-04-17:
Which version are you using? Did you installed it from CRAN? What is your operating system? I swear that `stopifnotlas` exists otherwise there are not a lot of functions that could  not work. I just tested it and i confirm it works. 

Maybe something went wrong during the installation. You may try to reinstall the package..

Indu-cherukuri, 2018-04-17:
Hi, Thanks for the quick revert. My operating system is Linux Mint. I installed it using install.packages("lidR") in RStudio from CRAN. I tried uninstalling the package and installing it again, but it didn't help. I am unable to figure out the issue.

Jean-Romain, 2018-04-17:
Well, I really embarrassed for you. I cannot reproduce this issue (Linux Mint too). Could you try the function `lasfilter`. This function call `stopifnotlas`. It is expected to fail on your computer. Does it?

bi0m3trics, 2018-04-17:
For what it’s worth, I seem to recall getting this message when I had a mismatch between lidR and rlas… like when I updateed/installed lidR from GitHub but was using an older version of rlas from CRAN. I don’t make that mistake anymore. ;)
A dump from sessionInfo() might help too…

Jean-Romain, 2018-04-18:
Great for you. I don't understand what happened. Hope this does not happen too often with other users...

closed 2018-04-18 by Jean-Romain