r-lidar/lidR #128: lastrees_silva boundary feature error.

bw4sz, 2018-05-01:
I believe there the silva implementation has an error that clumps together points on opposite sides of the tile. I've found this in a number of different tiles. It can be identified by looking at the convex hull of the treeID column. 

[image]

Reproducible code below, and attached pdf showing the error. Easier to look at the plots attached.

[silva_error.pdf](https://github.com/Jean-Romain/lidR/files/1964678/silva_error.pdf)

```

library(lidR)

# Read in data

LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las = readLAS(LASfile, select = "xyz", filter = "-drop_z_below 0")
col = pastel.colors(200)

# Compute canopy model

chm = grid_canopy(las, res = 0.5, subcircle = 0.3)
chm = as.raster(chm)
kernel = matrix(1,3,3)
chm = raster::focal(chm, w = kernel, fun = mean, na.rm = TRUE)

# Perform segmentation

ttops = lidR::tree_detection(chm, 5, 2)
crowns<-lidR::lastrees_silva(las, chm, ttops, max_cr_factor = 0.6, exclusion = 0.6,
                             extra = T)

# Get convex hulls
#These comes from a [package/analysis](https://github.com/weecology/TreeSegmentation) i'm working on.

get_convex_hulls<-function(tile,ID){
split_trees= split(tile@data, ID)
tree_polygons<-lapply(split_trees,convex_hull)

names(tree_polygons)<-NULL

#assign treeID as slot ID for each polygon
for(x in 1:length(tree_polygons)){
tree_polygons[[x]]@polygons[[1]]@ID<-names(split_trees)[x]
}

#bind into large SpatialPolygonsDataframe
convex_polygons<-do.call(raster::bind,unlist(tree_polygons))

#make into sp dataframe
IDs <- sapply(slot(convex_polygons, "polygons"), function(x) slot(x, "ID"))
df <- data.frame(ID=1:length(IDs), row.names=IDs)
result<-sp::SpatialPolygonsDataFrame(convex_polygons,df)
sp::proj4string(result)<-tile@crs
return(result)
}

convex_hull<-function(x,plot=FALSE){
ch<-grDevices::chull(x$X,x$Y)
poly_coords<-x[c(ch,ch[1]),c("X","Y")]
sp_poly <- sp::SpatialPolygons(list(sp::Polygons(list(sp::Polygon(poly_coords)), ID=1)))
return(sp_poly)
if(plot){
plot(sp_poly)
points(cbind(x$X,x$Y))
}
}

silva_convex<-get_convex_hulls(tile = las,ID = las@data$treeID)

#view convex hulls
raster::plot(silva_convex)

# Find strange polygons

#The largest ones should be them.

areaF<-c()
for(x in 1:length(silva_convex)){
areaF[x]<-silva_convex@polygons[[x]]@area
}

which.max(areaF)

# Confirm thats a weird polygon

raster::plot(silva_convex[254,])

# Is that confusion in the orignal tile or just an artifact of the convex hull?

split_trees= split(las@data, las@data$treeID)
original_data<-split_trees[[254]]

#Make sure that's the treeID
unique(original_data$treeID)

#find those points in the las
bad_points<-las%>% lasfilter(treeID==254)
raster::plot(extent(las))
points(cbind(bad_points@data$X,bad_points@data$Y),col='red')

# Its in the original segmentation. Those points all have the same treeID.
```

Thanks for your great work on this package, its been a pleasure to work with.

Jean-Romain, 2018-05-01:
Thank for reporting the bug. There is indeed something that went wrong. I'm investigating.

A more simple way to reproduce

```r
LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las = readLAS(LASfile, select = "xyz", filter = "-drop_z_below 0")
col = pastel.colors(200)

chm = grid_canopy(las, res = 0.5, subcircle = 0.3)
chm = as.raster(chm)
kernel = matrix(1,3,3)
chm = raster::focal(chm, w = kernel, fun = mean, na.rm = TRUE)

ttops = lidR::tree_detection(chm, 5, 2)
lidR::lastrees_silva(las, chm, ttops, max_cr_factor = 0.6, exclusion = 0.6)

plot(lasfilter(las, treeID == 254))
```

closed 2018-05-01 by Jean-Romain

Jean-Romain, 2018-05-01:
The bug has been fixed. I used back the `raster::extract`function internally instead of my own `fast_extract`. For now it is good enough but I have to fix my `fast_extract` function which is 15 times faster and much less memory greedy.

The branch `master` now contains version 1.5.0 of the package. Some visible change and possible incompatibilities: 
* `magrittr`is not loaded anymore. If you want to use the pipe `%>%` you need to load `magrittr`
* `catalog_options()` is deprecated.

bw4sz, 2018-05-01:
Thank you. Will pull in a moment.

Jean-Romain, 2018-05-14:
Might be of interest for @bw4sz @bi0m3trics @mxblsdl 

`lidR 1.6.0` (in development) has a new function `tree_hulls` that supports convex and concave hulls and returnd a `SpatialPolygonsDataFrame`.

```r
LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las = readLAS(LASfile, select = "xyz", filter = "-drop_z_below 0")

lastrees_li2(las, speed_up = 7)

convex_hulls  = tree_hulls(las)
concave_hulls = tree_hulls(las, "concave")

par(mfrow=c(1,2))
sp::plot(concave_hulls)
sp::plot(convex_hulls)
```

[image]

For convex hull it is 10 times faster than the code presented in the first post of this issue.

bw4sz, 2018-05-14:
> Hell yeah - leave it to you to make my code obsolete! ;) Seriously, that's
> awesome JR, and I got just the project to do a little bit of testing
> with...
>
> Sent from my Verizon 4G LTE smartphone

Thanks! will benchmark against my implementation as well.

commit c940a5cd (2018-05-01): Fix #128