r-lidar/lidR #129: Review the doc of v1.5.0

Jean-Romain, 2018-05-02:
- [ ] NEW.md
- [ ] class-lascatalog.r (from l30 and from l62)
- [ ] grid_canopy.r  grid_metrics.r grid_terrain.r grid_tincanopy.r (only the tag `@section` which is identical in each)  
- [ ] grid_catalog.r
- [ ] lasclip.r
- [ ] lasfilterdecimate.r
- [ ] lasfiltersurfacepoints.r
- [ ] lassmooth.r
- [ ] lastrees.r
- [ ] lastrees_li2.r
- [ ] lasupdateheader.r
- [ ] readLAS.r

irfanswat, 2018-05-06:
Hi Jean-Romain,
Thank you for the detailed package that covers almost everything related to LiDAR point cloud data. While reading through the different functions in lidR package, I came across the LAScatalog function. Indeed, it's an efficient way to deal with a bunch of point clouds; however, (according to my understanding) the catalogued lidar files are limited to a square (shape). In cases where users are interested in processing forest plot datasets, a circular shape will be required. In my workflow, I am using lidar point clouds extracted for individual circular field plots (n=~100). Using the current LAScatalog function creates a square boundary around the actually circular plot area. Is there any possibility, in LAScatalog, to use lidar point clouds that are circular and are treated as is? If not, probably either circular shapefiles can be used to extract point cloud data inside the shapefile or a convex hull can be created around the already clipped (with a circular shapefile) lidar point cloud data.

Many thanks,
Irfan

Jean-Romain, 2018-05-06:
I think `catalog_queries()` is what you are look for. A catalog allows to process the entire catalog in a continuous way or to extract regions of interest. `lasclip` also does this job since `v1.5.0`

irfanswat, 2018-05-10:
Many thanks Jean. I will get back to you if I had any query.

Cheers,