r-lidar/lidR #130: mosiac of .las objects?

bw4sz, 2018-05-02:
Is there a recommended way of combining .las tiles? I can see lasadddata, which seems designed for additional columns. I am trying to crop a .las by a SpatialPolygonsDataFrame. Unfortunately, the polygons overlap among multiple tiles. My goal is to first merge the underlying tiles and then crop.

bw4sz, 2018-05-02:
to clarify, what i mean is that if we have .las objects, could I rbind the @data slot from one to the another, and then recreate the header? Would that be acceptable?

bw4sz, 2018-05-02:
Sorry, not quite following.

Here is a catalog object.
```
> ctg
class       : LAScatalog
extent      : 394582.2 , 401000 , 3276265 , 3292022 (xmin, xmax, ymin, ymax)
area        : 52915190 m²
points      : 214972449 points
density     : 4.1 points/m²
num. files  : 66 
coord. ref. : +init=epsg:32617 +proj=utm +zone=17 +datum=WGS84 +units=m +no_defs +ellps=WGS84 +towgs84=0,0,0 
Processing options: 
 - split the dataset into 1000 x 1000 m tiles
 - each tile has a 15 m buffer
 - processing done using 1 core(s) if possible.
```

The docs say SpatialPolygonDataFrame, try that first. Here are a set of polygons.

```
> itcs[[1]]
class       : SpatialPolygonsDataFrame 
features    : 9 
extent      : 402390.1, 402416, 3286283, 3286317  (xmin, xmax, ymin, ymax)
coord. ref. : +proj=utm +zone=17 +datum=WGS84 +units=m +no_defs +ellps=WGS84 +towgs84=0,0,0 
variables   : 3
names       : crown_id, confidence,  Plot_ID 
min values  :       31,       High, OSBS_001 
max values  :      565,       High, OSBS_001 
```

```
> lasclip(ctg,itcs[[1]])
Error: Geometry not supported
```
or a single polygon

```
> itcs[[1]]@polygons[[1]]
An object of class "Polygons"
Slot "Polygons":
[[1]]
An object of class "Polygon"
Slot "labpt":
[1]  402401.3 3286294.9

Slot "area":
[1] 17.68021

Slot "hole":
[1] FALSE

Slot "ringDir":
[1] 1

Slot "coords":
          [,1]    [,2]
 [1,] 402402.5 3286298
 [2,] 402403.9 3286297
 [3,] 402403.4 3286294
 [4,] 402401.8 3286294
 [5,] 402401.2 3286292
 [6,] 402399.8 3286292
 [7,] 402398.6 3286293
 [8,] 402399.2 3286296
 [9,] 402402.5 3286298

Slot "plotOrder":
[1] 1

Slot "labpt":
[1]  402401.3 3286294.9

Slot "ID":
[1] "0"

Slot "area":
[1] 17.68021
```

```
> lasclip(x = ctg,geometry = itcs[[1]]@polygons[[1]])
Error: Geometry not supported
```

Or a SpatialPolygon, perhaps of the extent of the set.

```
> as(extent(itcs[[1]]),"SpatialPolygons")
class       : SpatialPolygons 
features    : 1 
extent      : 402390.1, 402416, 3286283, 3286317  (xmin, xmax, ymin, ymax)
coord. ref. : NA 
> f<-as(extent(itcs[[1]]),"SpatialPolygons")
> lasclip(ctg,f)
```

Version: lidR_1.5.0.9000

Jean-Romain, 2018-05-02:
I'm currently busy.  I think your problem is simple and has a simple solution. I give you an answer and a reproducible example tomorrow.

Jean-Romain, 2018-05-02:
With the `lidR` example dataset. Here I extract a lake (`SpatialPolygon`) from a `catalog`. The catalog is a single file `catalog` but it also works with a multi-file `catalog`.

Your error was to use a SpatialPolygon**s**

```r
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = catalog(LASfile)

shapefile_dir <- system.file("extdata", package = "lidR")
lakes = rgdal::readOGR(shapefile_dir, "lake_polygons_UTM17")

lake = lakes@polygons[[1]]@Polygons[[1]]
las = lasclip(ctg, lake)

plot(las)
```

bw4sz, 2018-05-02:
perfect!

closed 2018-05-02 by Jean-Romain

Jean-Romain, 2018-05-03:
Btw, the next time you have a question like that, which may be of interest for every users, I would prefer you to ask it on https://gis.stackexchange.com/ if possible. I would like github to be the place for reporting bugs and stackexchage the place to ask questions. This way, the questions will be better indexed by search engines and useful to everybody.

bw4sz, 2018-05-04:
> https://gis.stackexchange.com/questions/280511/clipping-
> object-of-class-las-with-polygon-shapefile-using-lidr

definitely.