r-lidar/lidR #132: Issue with Grid_metric function no longer finding PulseID

Selenachav, 2018-05-17:
Hello, I was running the below script in order to create voxels and also get a pulse count for the LAS file I was using it has been working fine but now I get an error message that says _'pulseID' not found_. Any help on fixing this so I can get a pulse counts again?

> grid_metrics3d(lidar1, length(Z),res = 0.5, debug = FALSE)
> 
> grid_metrics3d(lidar1, mean(ScanAngle),res = 0.5, debug = FALSE)
> 
> myMetrics = function(i, angle, pulseID)
> {
>   ret = list(
>     npulse  = length(unique(pulseID)),
>     angle   = mean(angle),
>     imean   = mean(i)
>   )
>   
>   return(ret)
> }
> 
> voxels1 = grid_metrics3d(lidar1, myMetrics(Intensity, ScanAngle, pulseID),res = 0.5, debug = FALSE)

**

Jean-Romain, 2018-05-17:
You are using the `lidR` version `1.5.0`

As said in the [NEWS](https://github.com/Jean-Romain/lidR/blob/master/NEWS.md):

> The option `P` is no longer set by default in `readLAS`.

Simply use `laspulse` or set the `P` flag in `readLAS`

Selenachav, 2018-05-17:
Would you be able to give me an example on how to set the pulse flag in the readlas function? I tried following the example on the wiki page but I get an error saying no P can't be found.
I also can't seem to use the laspulse function on the diffrent voxels layers I created.

Jean-Romain, 2018-05-17:
`readLAS("file.las", select = "*P")`

closed 2018-05-22 by Jean-Romain