r-lidar/lidR #133: Voxel operations

spono, 2018-05-22:
Ciao JR,
is there a way to subtract two voxelised scenes (e.g. time1-time2) or may it be a possibility for enhancement?

Jean-Romain, 2018-05-22:
Hi, the output of `grid_metrics3d` is a `data.table`. What you want to do can be achieved in R base or with `dplyr` or more efficiently with `data.table` capabilities. What you are looking for is a `merge`  by `X` `Y` and `Z` keys.

https://rstudio-pubs-static.s3.amazonaws.com/52230_5ae0d25125b544caab32f75f0360e775.html

spono, 2018-05-22:
I might have solved in this way  

`# create uid from XYZ`
  `data1[,uid := paste0(X, Y, Z)]`
  `data2[,uid := paste0(X, Y, Z)]`
  
  `# set uid as search key`
  `setkey(data1, cols = "uid")`
  `setkey(data2, cols = "uid")`
  
  `# identify voxels unique to data1`
  `out = data1[!data2]`

thanks!
  `out[,uid := NULL]`

Jean-Romain, 2018-05-22:
This is a workable solution :wink:

closed 2018-05-22 by Jean-Romain