r-lidar/lidR #135: grid_catalog returns a table with no class

Jean-Romain, 2018-05-23:
After processing a `LAScatalog` with `grid_catalog` the output was a `data.table` but the class was `NULL` i.e. R was not able to recognize the structure.

closed 2018-05-23 by Jean-Romain

spono, 2018-06-07:
ciao JR,
I still get this bug using `grid_canopy` on the `master` release. If i load the single file as a `las` its works, while if I load it as a `catalog` it doesn't.
Unfortunately [like always] I don't get this error on every file. Any hint where to start to check for errors?

Jean-Romain, 2018-06-08:
You can patch it easily by assigning a class to your table

```r
as.lasmetrics(output, resolution)
```

The bug come from the fact that, when processing a catalog, some cluster can be empty (no data). If so, `NULL` is returned. This is a normal behavior, but the first cluster is used to find the class of the output. If it is `NULL` then the output class turns to `NULL`.

This bug is expected to be fixed in 1.5.1. I'm suprised by your issue. Is it really the same?

spono, 2018-06-11:
the behaviour is the same but I don't know if the origin of the issue is the same. I'll try to test some more during this week.

commit 88269134 (2018-05-23): Fix #135