r-lidar/lidR #138: lasground

spono, 2018-06-04:
Ciao JR,
does `lasground` work on all points or it automatically subsets only 'last and single' returns? I think that such behaviour would allow to speed up the process and avoid to introduce points from different return number. Am I wrong?

Jean-Romain, 2018-06-04:
You're right `lasground` use all the point you provided as input. But I think that dropping  non-last-or-single returns won't really speed-up the algorithm. But you can bechmark it:

```r
las1 = readLAS(...)
las2  = lasfilter(las1, (ReturnNumber == 1 & NumberOfReturn == 1) | (ReturnNumber == NumberOfReturn)

system.time({lasground(las1, ... )})
system.time({lasground(las2, ... )})
```

Please tell me for different configuration if your results are significant. If so I will add this filter at the C++ level in `lasground`. This way it will be memory efficient.

spono, 2018-06-04:
`ws = c(1,2,3,4,5)`
`th = seq(0.1, 0.5, length.out = length(ws_par))`

here the results:
las = 11mln points     -> elapsed 1742.92
las2 = 8,2mln points  -> elapsed 1101.06

I don't know if the difference is significant but I think that updating only the last&single points and leaving the others unchanged could be, if not faster, at least "safer" (from a point of view of correctness of the classified points).

Jean-Romain, 2018-06-04:
Added in my todo list for `lidR 1.6.0`

spono, 2018-06-06:
great.
maybe implemented as parameter `all_points = FALSE` to let the maximum flexibility for other kind of elaborations.

Jean-Romain, 2018-06-06:
Exactly what I planned to do :smile:

Jean-Romain, 2018-07-11:
Done :tada:

closed 2018-07-11 by Jean-Romain