r-lidar/lidR #140: catalog(): Can you export the ctg:  LAScatalog to a geojson/shapefile/....

pieterz, 2018-06-06:
Hey!

I dont know if this is the right place to post this (sorry in advance), but can you export the las/laz tile index (ctg: Large LAScatalog) that is created with the function catalog('_folder'_) to a geojson or shapefile in R? 

I would like to make a clickable map of a folder with mutiple lazfiles (so you can download the laz tiles from a website), however i am struggling finding a function that can create a laz tile index (with directory to the files) and the function catalog() seems to make an excellent index. Also really like the package!

Hopefully you can help me.

Thanks for your time!

Kind regards,

Pieter

bi0m3trics, 2018-06-06:
to quote JR - "if your question is pertinent and of interest for everybody. I would prefer you to ask questions on https://gis.stackexchange.com/ (if possible)." He usually goes on to say that "github issues are not very well indexed by search engines so basically my answer is useful for you only" and that he "would like github to be the place for bug reports and stackexchange the place for questions. That way other people would be able to answer"

Now to your question - if I understand you correctly, it's as easy as a call to rgdal::writeOGR after coercing the catalog to a spatial object:
`project = catalog("path to las/z files")`
`rgdal::writeOGR(as.spatial(project), "C:/path/project.shp","layer_name", driver="ESRI Shapefile") # for a shapefile on windows or`
`rgdal::writeOGR(as.spatial(project), "C:/path/project.shp","layer_name", driver="GeoJSON") # for a geojson`

pieterz, 2018-06-06:
>  I would prefer you to ask questions on https://gis.stackexchange.com/ (if possible).
Ah thanks for the tip! Didn't know github issues weren't very well indexed.

> Now to your question - if I understand you correctly
Yes this was exactly what I was looking for. Thank you very much!

Kind regards,

Pieter

Jean-Romain, 2018-06-06:
Thanks @bi0m3trics .

closed 2018-06-06 by Jean-Romain

Hannes-Ole, 2021-09-22:
Although I agree on the github<>stackoverflow part, ironically, I found this as the (apparently well-indexed) first entry of a google-search and it helped me 👍 thank you