r-lidar/lidR #144: R version problem

swinersha, 2018-06-12:
I'm trying to install lidR on R versions 3.4.0, 3.4.3 and 3.4.4. I've tried using CRAN and through Github via devtools with no success. Which version of R do I need to install lidR at the moment? If lidR currently only works with older versions of R it would be helpful if you could make this information available in the readme so that people can get up and running quickly. 

Thanks

Tom

Jean-Romain, 2018-06-12:
`lidR` works on all the R versions. 

### CRAN

At the time being `lidR` has been removed from the CRAN because of an issue with `rlas` that is all my bad :worried: . I expect to fix it within few days. I'm sorry.

### Github

You basically said me *it does not work*. That is not really helpful. It should work. What is going wrong?

swinersha, 2018-06-12:
Yes. Sorry that was unclear.  Thanks for the rapid response.

I was getting the errors like "not possible to uninstall previous version of data.table" (not verbatim) and the same for Rcpp. 

I just updated my devtools to the version built under 3.4.4 and installed rlas and lidr with no problems. 

Thanks again.

Jean-Romain, 2018-06-12:
> not possible to uninstall previous version of data.table

This is not related to `lidR` but to `Rcpp` and `data.table`. What if you install these packages manually

```r
install.packages(c("Rcpp", "data.table"))
```

swinersha, 2018-06-12:
That might work. I have updated devtools now and everything worked fine so no problem as far as I'm concerned.

closed 2018-06-12 by Jean-Romain