r-lidar/lidR #146: never-ending segmentation

spono, 2018-06-22:
Ciao JR,
using `lastrees_dalponte` I get a segmentation that doesn't conclude. 

r = lastrees_dalponte(impLAS, chm=r, treetops=ttops, max_cr=20, extra = TRUE)

this happens on a specific plot, but I tested the other chm-based algs and they work fine.

Jean-Romain, 2018-06-22:
Well, I will need the specific plot to reproduce.

Jean-Romain, 2018-06-22:
Thank for sharing a reproducible example by email. I reproduced the bug. First I added a way to exit the infinite loop with the stop button. Now I will try to figure out what happened.

Jean-Romain, 2018-06-22:
I think I got the issue. You have tree tops that are not actually tree tops. Thus, I don't know exactly where in the code, the algorithm failed with region growing. I have to figure out were does it come from to fix that but you also need on your side to use correct seeds.

This works well

```r
# set project variables
wdir = file.path("~/Téléchargements/","bug_146")

projectCRS="+proj=tmerc +lat_0=0 +lon_0=15 +k=0.9999 +x_0=500000 +y_0=0 +ellps=bessel +towgs84=682,-203,480,0,0,0,0 +units=m +no_defs"

crown.rad = 5 # estimated average crown radius

rast.resol = 50 # resolution for output rasters; in cm

library("lidR")
library(raster)

# Load the data
impLAS = readLAS(file.path(wdir,"plot_norm.laz"), select="* P")
impLAS@crs <- sp::CRS(projectCRS)

plot(impLAS)

# CHM
r = grid_canopy(impLAS, res=rast.resol/100)
r = as.raster(r)
kernel = matrix(1,3,3)
r = raster::focal(r, w = kernel, fun = mean, na.rm = TRUE)

plot(r, col = height.colors(50))

ttops = tree_detection(r, 9, 2)

plot(ttops, add = T, col = "black")

crown = lastrees_dalponte(impLAS, r, ttops, max_cr=2*crown.rad/(rast.resol/100), extra = TRUE)

plot(las, color = "treeID", colorPalette = col)
```

Jean-Romain, 2018-06-23:
I got it. Your CHM raster is not the same size than your tree tops raster. That introduced an index error when reading the matrix.

```r
> ttops
class       : RasterLayer 
dimensions  : 72, 72, 5184  (nrow, ncol, ncell)
resolution  : 0.5, 0.5  (x, y)
extent      : 458572, 458608, 58948, 58984  (xmin, xmax, ymin, ymax)
coord. ref. : +proj=tmerc +lat_0=0 +lon_0=15 +k=0.9999 +x_0=500000 +y_0=0 +ellps=bessel +towgs84=682,-203,480,0,0,0,0 +units=m +no_defs 
data source : /home/jr/Téléchargements/bug_146/ttops.tif 
names       : ttops 
values      : 1, 17  (min, max)

> r
class       : RasterLayer 
dimensions  : 73, 72, 5256  (nrow, ncol, ncell)
resolution  : 0.5, 0.5  (x, y)
extent      : 458572, 458608, 58947.5, 58984  (xmin, xmax, ymin, ymax)
coord. ref. : +proj=tmerc +lat_0=0 +lon_0=15 +k=0.9999 +x_0=500000 +y_0=0 +ellps=bessel +towgs84=682,-203,480,0,0,0,0 +units=m +no_defs 
data source : in memory
names       : Z 
values      : 0.01001221, 33.28001  (min, max)
```

This was not expected to work because raster matching is checked before to enter in the C++ code. However I found a weird behavior with the `!=` operator on `extent`

```r
> raster::extent(chm) != raster::extent(treetops)
[1] FALSE
> raster::extent(chm) == raster::extent(treetops)
[1] FALSE
```

Thus the correct way to compare the extent is 

```r
!raster::extent(chm) == raster::extent(treetops)
```

Now the code behave as expected, it fails with a message error

> chm and treetops do not match together

You can fix that by cropping your CHM

```r
chm = raster::crop(r, ttops)
crown = lastrees_dalponte(impLAS, chm, ttops, max_cr=2*crown.rad/(rast.resol/100), extra = TRUE)
```

However, your treetops are not tree tops and you CHM not smoothed. So your segmentation is just awful.

spono, 2018-06-25:
ahahah
awful but at least I hope it will be for good reasons ;)

btw, it now works: thank you very much

closed 2018-06-25 by Jean-Romain

Jean-Romain, 2018-06-26:
In version 1.6.0 the matching is more robust. Seeds and CHM are matched by coordinates and not by cells. Thus input are not expected to be the same size or even the same resolution. If one seed is outside the CHM the method stops.