r-lidar/lidR #149: catalog_apply stops

biometrica, 2018-07-03:
I've been working with lidR using data which is not for public distribution.  Unfortunately, that means I cannot create a reproducible error for inspection.  I've tried calling catalog_apply for normalization of las files, covering around 10k acres.  The algorithm stops abruptly at different clusters, with varying size, after about 1 minute of processing.  The same data can be manually processed when I load and normalize the las file by hand.  I tried using traceback() without much success to debug.

Its hard to say what is hanging up the apply function, but some more diagnostics for when stop occurs could help.  Any ideas?

Error message reads as follows:
Progress: 31%Error in interpolate(las@data[Classification == 2, .(X, Y, Z)], las@data[,  : 
  argument "method" is missing, with no default

Jean-Romain, 2018-07-03:
Hi, if you can't share the data, at least please share the code.

biometrica, 2018-07-03:
dirlas = paste0(dirdat, "/las")
setwd(dir)

# packages
require(lidR)
require(lattice)
require(rgdal)

# Plot the extent of a given .las files
prj = catalog(dirlas)

# Plot the extent of the area
plot(prj)

# Vary tiling size
cores(prj) = 30
tiling_size(prj) = 1000

catalog_apply(
  ctg = prj,
  lasnormalize,
  method = "knnidw"
)

Jean-Romain, I've tried varying the cores, tile size and method with no good results.  Initially, I assumed there was an issue in the .las file.  However, by varying the settings, I found the apply function would stop at different panels of the .las files.  Perhaps I've misread the documentation for the function?  Again, I can normalize these .las files using a for loop, but it takes a considerable amount of time for clipping and iteration.

Jean-Romain, 2018-07-03:
First you misused `catalog_apply`. It is partly my bad because the function is not really well designed (I cannot go back without creating version incompatibilities) but in an other hand it is well documented. That will be better:

```r
args = list(method = "knnidw")
catalog_apply(ctg = prj, lasnormalize, func_args = args)
```

That being said, the function is useless like that because, yes the normalization is done, but is not saved anywhere. Something like that can be created

```r
myfunc = function(las)
{
  file = "generate/a/file/name.las"
  lasnormalize(las, method = "knnidw")
  write(las, file)
}

catalog_apply(prj, myfunc)
```

bi0m3trics, 2018-07-03:
Looks like JR beat me to it and he’s probably right but I had already types this up so here goes:

Since you’re specifying tiling_size (and varying them) which are likely different from your actual las tile file sizes, the problem may also be the clustering of catalog_apply is creating a cluster (or tile) that has no data in it and thus lasnormalize has nothing to work with.. I wonder if adding a “if(is.null(las)) return(NULL)” before the call to lasnormalize would help…

For example:
project = catalog(“path to files”)
normalize.tiles = function(las)
{
if (is.null(las))
    return(NULL)
lasnormalize(las, method = "knnidw", k = 10L)
# do something with the normalized las and assign it to m
return(m)
}
output = catalog_apply(project, normalize.tiles, select = "xyzinrc")

Jean-Romain, 2018-07-03:
@bi0m3trics 

>  I wonder if adding a “if(is.null(las)) 

Right ! but in `v1.6.0` this is now managed internally, the user don't have to think about this case anymore.

biometrica, 2018-07-03:
I understand the apply process now.  Thanks for clarifying Andrew and Jean-Romain.  Andrew correctly identified the probable issue.  When the algorithm moves over a large extent, it eventually hits an empty patch with no points and blows up.  The debugging process for the apply function is enhanced by Andrew's suggestion to write out intermediate outputs.   The documentation is very complete Jean-Romain.  For a new user, it is a lot to take in, especially when you are working with new data (or are new to LiDAR in general).  A vignette with more examples using lazyData or an introduction to the package could be very powerful in getting new users up to speed.

closed 2018-07-03 by biometrica

bi0m3trics, 2018-07-03:
I'm all about not thinking! ;)

Jean-Romain, 2018-07-03:
You can start here:

https://github.com/Jean-Romain/lidR/wiki

And this page is very extensive too:

https://github.com/Jean-Romain/lidR/wiki/Apply-user-defined-analyses-over-an-entire-catalog

There are a lot of resources, it took me days of work to write all that stuff but I don't have enough free time to compile that in a set of official vignettes. I plan to use some wiki pages as vignette in `1.6.0` but not an overall vignette about the package. Any PR is welcome.

Also basically all the code for `LAScatalog` stuff should be rewritten from scratch. Many user want these features that were initially added for some specific things. But the package being more and more widely used users want to process whole catalogs in a straightforward way. It is has not been designed like that. I'm thinking about a `lidR v2.0` with strong foundations more than about vignettes :wink:

bi0m3trics, 2018-07-03:
“lidR v2.0 with strong foundations” – completely understandable. I’ve got a couple ideas we can discuss off github… more form me later.