r-lidar/lidR #150: Discrepancy in help for lasclip

biometrica, 2018-07-03:
The help file for lasclip indicates: geometry = a geometric object. Currently Polygon and SpatialPolygonsDataFrame from sp are supported and x = An object of class LAS or LAScatalog.  

_However, when I input a polygon with class:_
[1] "SpatialPolygonsDataFrame"
attr(,"package")
[1] "sp"

_As follows,_
  # Clip to extent
  las = lasclip(
    x = prj,
    geometry = stand
  )

Where, prj is a catalog with class:
[1] "LAScatalog"
attr(,"package")
[1] "lidR

_The following error is produced:_
Error: 'SpatialPolygonsDataFrame' is not supported for 'LAScatalog' objects

Jean-Romain, 2018-07-03:
Hi

The documentation is pretty clear on that subject:

> Minor differences exist between the LAS and LAScatalog version of lasclip. For example the user can clip a SpatialPolygonsDataFrame from a LAS object but not from a LAScatalog. Also the option inside = FALSE is disabled for LAScatalog objects. These differences are generally justified by memory safety concerns.

You can't clip a `SpatialPolygonDataFrame` from a `LAScatalog`.  There are several reasons for that. Some of them can be found is this thread: https://gis.stackexchange.com/questions/280511/clipping-object-of-class-las-with-polygon-shapefile-using-lidr

biometrica, 2018-07-03:
Thanks for clarifying that for me, Jean-Romain.  I've found the package to be very useful.  It would be nice if a workshop or webinar were offered at some point for new users.

closed 2018-07-03 by biometrica

Jean-Romain, 2018-07-03:
Please invite me for a workshop and I will be glad to come :wink: