r-lidar/lidR #154: grid_terrain function is failing

komazsofi, 2018-07-17:
Dear Jean-Romain,

I got some error message during I am using the grid_terrain function. The error message is the following:

Error in C_points_in_polygon(hull[, 1], hull[, 2], grid$X, grid$Y) : 
  object '_lidR_C_points_in_polygon' not found

My code:

# Import required libraries
library("lidR")
library("raster")

# Import data
las = readLAS("lauwermeer_merged.las")

##########################
# Preprocess - Normalize #
##########################

las_ground = lasfilter(las, Classification == 2)
plot(las_ground)

dtm = grid_terrain(las_ground, method = "knnidw", res=5)

I am not sure where this error is coming from. The las file is a normal las file with Classification flag where 2 represents the ground points. I am using 1.6.0 version on a Windows 10 machine. 

By the way really nice package!

Best,
Zsofia

Jean-Romain, 2018-07-17:
Hi! This is a known issue. When compiling the package from source on Windows many user encountered missing functions. This does not come from the source but from the magic of Windows :unamused:

Hopefully I compiled a version 1.6.0 for windows on CRAN yesterday and it is still available for 48h. You can download and install the binaries for windows: https://win-builder.r-project.org/BNijwNTuG07X/

Jean-Romain, 2018-07-17:
That one is more recent (available 72h) https://win-builder.r-project.org/m67auW7DQ9ZX/

closed 2018-07-17 by Jean-Romain

komazsofi, 2018-07-19:
Thank you it is working!