r-lidar/lidR #155: Error "Could not find function" when using catalog_apply

frank2165, 2018-07-19:
Hello, the following issue occurred when I moved to the most recent version of `lidR` and did not seem to occur in version 1.4.2. The issue is mostly caused by the future package but has a work-around in the lidR package that may (or may not) be worth implementing.

When I supply a function to catalog_apply that contains functions from other packages, e.g. magrittr's pipe operator, I get an error which states that `%>%` could not be found. I could not determine the reason why this happens but I suspect that it has some thing to do with how future finds globals and packages. The following (highly trivial) example should hopefully reproduce the error:

```
library(lidR)
library(magrittr)
voxel_metrics <- function(las){
  las %>% grid_metrics3d(las, func = list(Count = length(Intensity)))
}
voxels <- catalog_apply(my.catalog, func = voxel_metrics)
```

The issue can be resolved simply enough by using the `::` operator to make the call to function in a different package more explicit (and not using magrittr at all). But I found that the problem goes away if I butcher the nice clean code in the catalog_apply function as follows.

```
catalog_apply <- function(ctg, func, func_args = NULL, ...)
{
  assertive::assert_is_all_of(ctg, "LAScatalog")
  assertive::assert_is_function(func)

  progress  <- progress(ctg)
  ncores    <- cores(ctg)
  stopearly <- stop_early(ctg)
  clusters  <- catalog_makecluster(ctg, 1)
  f <- function(cluster) { 
      las <- readLAS(cluster, ...)
      if (is.null(las)) return(NULL)
      return(do.call(func, c(las, func_args)))
  }
  output <- cluster_apply(clusters, f, ncores, progress, stopearly)
  # output  <- cluster_apply(clusters, cluster_apply_func, ncores, progress, stopearly,
  #                          func = func, ctg = ctg, func_args = func_args, ...)
  return(output)
}
```

frank2165, 2018-07-20:
Thank you for looking into this. I am indeed working on Windows, sorry I forgot to mention that, so I believe future is defaulting to `plan("multisession")`. If I can be of any help with testing just let me know.

Jean-Romain, 2018-07-21:
@frank2165 I tried with `plan(multiprocess)`, `plan(multisession)` and `plan(multicore)` and I did not reproduce the error. Could you help me to reproduce.

frank2165, 2018-07-21:
Hello again, thanks for taking the time to have a look into this.

I have run the example using a separate data set on my Windows machine at home and continue to get the error. After stepping through the execution of `future::future` I'm not sure why the bug is platform dependent, it's my (very limited) understanding that `future::multisession` just uses the parallel package to create the required number of fresh R sessions to run on different processors (which doesn't require forking the main R process and so should work in the same fashion for both linux and Windows).

From what I have been able to gather from stepping through the execution of future, the error is occurring because of the way future searches for packages/names to send to the worker processes. Actually that's not quite accurate, `future::getGlobalsAndPackages` calls `globals::globalsOf` which uses the codetools package to search for packages/names in expressions. To elaborate further, `getGlobalsAndPackages` looks inside `cluster_apply_func` and finds a bunch of primative/base package functions as well as the function `readLAS`. Since `readLAS` belongs to the lidR namespace it exports the lidR package to the cluster. The `getGlobalsAndPackages` function also finds a bunch of variables, such as `func`, `func_args` and `...` and exports these as well, it just doesn't seem to do any analysis of `func` (the function supplied to catalog_apply).

I have included the debug information provided by the future package when using the current version of the lidR package (catalog_apply_test_master.txt) and the same trace obtained after modifying the package as described in the original comment (catalog_apply_test_modified.txt). This was the only information I could think of that would help, but if there's some other output that you require I'll do my best to provide that too.

If you'd like to see the changes I made to catalog_apply.R and cluster_apply.R I can provide them in a pull request.

I hope this helps.

[catalog_apply_test_lidR_master.txt](https://github.com/Jean-Romain/lidR/files/2216408/catalog_apply_test_lidR_master.txt)
[catalog_apply_test_lidR_modified.txt](https://github.com/Jean-Romain/lidR/files/2216409/catalog_apply_test_lidR_modified.txt)

Jean-Romain, 2018-07-21:
Ok, my bad, I reproduced the error. I simply forgot to set a number of core > 1 the first time. I think I will make an R base reproducible example and open an issue to the `future` package.

Jean-Romain, 2018-07-21:
I can see thee options:

User-controlled options

```r
voxel_metrics <- function(las)
{
  library(magrittr)
  las %>% grid_metrics3d(func = list(Count = length(Intensity)))
}
```

Full automatic overkilled option and probably not that much robust

```r
output[[i]] <- future::future({ f(clusters[[i]], ...) }, substitute = TRUE, packages = names(sessionInfo()$otherPkgs))
```

Another user-controlled options

```r
voxel_metrics <- function(las) {
  las %>% grid_metrics3d(func = list(Count = length(Intensity)))
}

voxels <- catalog_apply(my.catalog, func = voxel_metrics, extra_package = c("magrittr"))

frank2165, 2018-07-22:
Since the problem seems to result from the user supplied function not being analysed for globals/packages to export another option is to add the following code to `cluster_apply`, just below the call to `plan("multiprocess")`.

```r
dots <- list(...)
is.fun <- vapply(dots, is.function, logical(1))
required.pkgs <- "lidR"
if(any(is.fun)){
  dots <- dots[is.fun]
  for(fun in dots){
    globals <- future::getGlobalsAndPackages(fun)
    where   <- attr(globals$globals, "where")
    pkgs    <- vapply(where, attr, character(1), "name", USE.NAMES = FALSE)
    pkgs    <- gsub("package\\:", "", pkgs)
    required.pkgs <- c(required.pkgs, pkgs)
  }
```

The required packages are then passed onto future

```r
output[[i]] <- future::future({ f(clusters[[i]], ...) }, substitute = TRUE, packages = required.pkgs)
```

Jean-Romain, 2018-07-22:
That's definitively the most elegant option. Thanks for this contribution.

commit fa83de87 (2018-07-22): Fix #155
closed 2018-07-22 by Jean-Romain