r-lidar/lidR #157: Error: File(s) not supported

FalcoNYCDEP, 2018-07-20:
Hello, 

I am having issues with a set of .las files. lidR does not support them.  The original files were v1.1 and I updated to v1.4 using 'las2las' with the LAStools application from rapidlasso.  Is there something odd about my files that sticks out to you?  

Thank you! 

**This is the info on one of my original las files:** 

lasinfo -i "C:\Users\gfalco\Documents\R\lidR\las\u_4770066125_2009.las"
lasinfo (180620) report for 'C:\Users\gfalco\Documents\R\lidR\las\u_4770066125_2009.las'
reporting all LAS header entries:
  file signature:             'LASF'
  file source ID:             0
  global_encoding:            0
  project ID GUID data 1-4:   00000000-0000-0000-0000-000000000000
  version major.minor:        1.1
  system identifier:          ''
  generating software:        'TerraScan'
  file creation day/year:     0/0
  header size:                227
  offset to point data:       229
  number var. length records: 0
  point data format:          1
  point data record length:   28
  number of point records:    943948
  number of points by return: 633163 256081 49752 4952 0
  scale factor x y z:         0.01 0.01 0.01
  offset x y z:               0 0 0
  min x y z:                  477000.01 4661250.00 341.21
  max x y z:                  477750.00 4661999.99 397.96
the header is followed by 2 user-defined bytes
reporting minimum and maximum for all LAS point record entries ...
  X            47700001   47775000
  Y           466125000  466199999
  Z               34121      39796
  intensity           0        255
  return_number       1          4
  number_of_returns   1          4
  edge_of_flight_line 0          1
  scan_direction_flag 0          1
  classification      1          7
  scan_angle_rank   -20         20
  user_data           0          0
  point_source_ID   422        577
  gps_time 317029.380465 419981.856851
number of first returns:        633163
number of intermediate returns: 60037
number of last returns:         624572
number of single returns:       373824
overview over number of returns of given pulse: 373824 405498 144536 20090 0 0 0
histogram of classification of points:
          717697  unclassified (1)
          223173  ground (2)
            3078  noise (7)

**Here is the same report after converting to v1.4 using rapidlasso LAStools:**

lasinfo (180620) report for 'C:\Users\gfalco\Documents\R\lidR\las_updated\u_4770066125_2009.las'
reporting all LAS header entries:
  file signature:             'LASF'
  file source ID:             0
  global_encoding:            0
  project ID GUID data 1-4:   00000000-0000-0000-0000-000000000000
  version major.minor:        1.4
  system identifier:          'LAStools (c) by rapidlasso GmbH'
  generating software:        'las2las (version 180620)'
  file creation day/year:     0/0
  header size:                375
  offset to point data:       377
  number var. length records: 0
  point data format:          1
  point data record length:   28
  number of point records:    943948
  number of points by return: 633163 256081 49752 4952 0
  scale factor x y z:         0.01 0.01 0.01
  offset x y z:               0 0 0
  min x y z:                  477000.01 4661250.00 341.21
  max x y z:                  477750.00 4661999.99 397.96
  start of waveform data packet record: 0
  start of first extended variable length record: 0
  number of extended_variable length records: 0
  extended number of point records: 943948
  extended number of points by return: 633163 256081 49752 4952 0 0 0 0 0 0 0 0 0 0 0
the header is followed by 2 user-defined bytes
reporting minimum and maximum for all LAS point record entries ...
  X            47700001   47775000
  Y           466125000  466199999
  Z               34121      39796
  intensity           0        255
  return_number       1          4
  number_of_returns   1          4
  edge_of_flight_line 0          1
  scan_direction_flag 0          1
  classification      1          7
  scan_angle_rank   -20         20
  user_data           0          0
  point_source_ID   422        577
  gps_time 317029.380465 419981.856851
number of first returns:        633163
number of intermediate returns: 60037
number of last returns:         624572
number of single returns:       373824
overview over extended number of returns of given pulse: 373824 405498 144536 20090 0 0 0 0 0 0 0 0 0 0 0
histogram of classification of points:
          717697  unclassified (1)
          223173  ground (2)
            3078  noise (7)

FalcoNYCDEP, 2018-07-20:
Thank you, I am running lidR v1.5.0 & rlas v1.2.3. I'll have to wait until Monday to get a clean install on my server and will let you know if that fixes it.  Have a nice weekend.

Jean-Romain, 2018-07-20:
Hi! This is an issue for the [rlas](https://github.com/Jean-Romain/rlas) package. Please open an issue on the `rlas` repo with a las file version 1.4 (or a piece of file) in attachment. Thanks. I will try to enable the support of file 1.4.

closed 2018-07-20 by Jean-Romain