r-lidar/lidR #160: Question: mutually exclusive tree_hulls

biometrica, 2018-07-30:
@Jean-Romain Is there a way to ensure the objects created using tree_hulls are mutually exclusive? 

Thanks for the updates to the latest lidR package and detailed release notes.

Jean-Romain, 2018-07-30:
Do you mean that the hulls do not intersect each other ? With concave hulls the intersections are likely to be null most of the time. With convex hull the polygons will be in intersection each other almost always.  Btw I can't see any method that could allows to make always mutually exclusive hulls.

biometrica, 2018-07-30:
Correct, JR.  I would like the tree_hull algorithm to be constrained so the hulls must also be mutually exclusive (i.e. they don't intersect).  That makes the math a bit easier when creating a confusion matrix of tree detection rate.  For example, consider calculating probability of number of trees in hulls: P(count = 0), P(count = 1), ... , P(count = n).  The math becomes difficult when you assign a tree into two hulls (i.e. hulls overlap the location of one tree), for example, when comparing stem mapped plots and the hulled LiDAR point cloud.  Does that make sense?

Jean-Romain, 2018-07-30:
Well it is a complex problem. What kind of technical solution are you expecting. If you have a name of an algorithm you can suggest something. But on my side I'm not sure your problem have a solution.

biometrica, 2018-07-30:
Agreed.  I don't know of an off the shelf solution.  Thanks for exposing your code on Git.  Perhaps I will write something to address this point.  Thanks for the quick response and confirming my understanding of the hull algorithm.

closed 2018-07-30 by biometrica