r-lidar/lidR #164: area() isn't working

ervanSTRI, 2018-08-20:
Dear Jean-Romain,

While working with your package and trying to retrieve the crown area of individual trees, I got the following message (in bold below). Area is computed internally, but but seems not to be accessible through the area() function...

 A <- lasfilter(MNS09,treeID==1)  # subsetting a tree after Li's point cloud classification
> A
class        : LAS (LASF v1.2)
memory       : 30.6 Kb 
extent       : 2504093 , 2504100 , 1119006 , 1119015 (xmin, xmax, ymin, ymax)
area         : 41.8 m² (convex hull)
points       : 388 points,  NA pulses
density      : 9.28 points/m²,  NA pulses/m²
field names  : X Y Z Intensity ReturnNumber NumberOfReturns Classification ScanAngle Zref treeID 
coord. ref.  : NA 

> area(A)
**Error in (function (classes, fdef, mtable)  : 
  unable to find an inherited method for function ‘area’ for signature ‘"LAS"’**

Same with your example:
> LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
> las = readLAS(LASfile)
> area(las)
Error in (function (classes, fdef, mtable)  : 
  unable to find an inherited method for function ‘area’ for signature ‘"LAS"’

Any clue?
Thanks!!
Ervan

Jean-Romain, 2018-08-20:
`area` works on my computer. Please report your package version and operating system.

Anyway you should used the function `tree_metrics` to achieve what you are trying to do :wink:

Jean-Romain, 2018-08-20:
`tree_hulls` actually

ervanSTRI, 2018-08-20:
Hi JR,

Thanks for your prompt reply. I have an outdated version of your package (1.4.0)

closed 2018-08-20 by ervanSTRI