r-lidar/lidR #166: LAScatalog not exported

AndyPL22, 2018-08-22:
I am writing a package in which I define the following class union:

`setClassUnion('CloudOrNull', c('LAScatalog', 'NULL'))`

I have added the `@importClassesFrom lidR LAS` tag to ensure the `LAS` class is imported.

When I build the package, I get the following warning message

    Warning message:
    class "LAScatalog" is defined (with package slot 'lidR') but no metadata object found to revise superClass information---not exported?  Making a copy in package '[my package]'

I looked up this issue, which has been explored thoroughly here:
https://stackoverflow.com/questions/28871632/how-to-resolve-warning-messages-metadata-object-not-found-spatiallinesnull-cla

It sounds like the easiest solution is simply to have `LAScatalog` exported. I was wondering if this would cause any issues?

Thanks!

Jean-Romain, 2018-08-22:
Hi,

To be honest I don't have any idea of the origin of your issue. The topic on stack overflow is long and complex I need time to read and understand it correctly.

But before to figure out what all that means if you want to build something on top of lidR you must rely on [lidR v2.0](https://github.com/Jean-Romain/lidR-v2.0). This version is available [here](https://github.com/Jean-Romain/lidR/tree/v2). It is currently an early draft but it works. Many details are prone to change but now a `LAScatalog` inherit from a `SpatialPolygonsDataFrame` and that will not change. It also have 3 others slots that will may be changed. Being a `SpatialPolygonsDataFrame` maybe your issue will be solved. Otherwise, if you have an idea of how to solve this problem please make a PR.

AndyPL22, 2018-08-22:
Sounds good to me. Thanks!

Jean-Romain, 2018-08-24:
I read the stackoverflow thread and not everyting is clear to be. However I made a toy package to reproduce your error. Actually the issue was pretty simple. The  documentation stated

```r
#' @exportClass LAS
```

Instead of

```r
#' @exportClass LAScatalog
```

`LAScatalog` (and `LAS`) are now exported (`v2.0.0`) and it seems to work.

AndyPL22, 2018-08-24:
Beautiful. Thanks for making the change!

closed 2018-08-24 by Jean-Romain