r-lidar/lidR #167: lastrees_dalponte error: Duplicated seed IDs

Jack-Hastings, 2018-08-23:
Hello,
I've been successfully using lastrees_dalponte for several weeks now. However, today I went to tweak some of the parameters and I received the error: Duplicated seed IDs. I'd never received the error in the past, and I did not change the inputs (LAS file, CHM, treetops). 

I was reading through the code and found this line:

if (length(unique(treetops_df[[3(]])) != nrow(treetops_df))
      stop("Duplicated seed IDs.", call. = FALSE)

It seems that this is saying that if the treetop height ($Z) is not unique than the function will fail. 
For example, I have: 
length(unique(treetop[[3]])) = 810
nrow(treetop) = 1359

Am I understanding the error correctly? 

Thank you,
Jack

Jean-Romain, 2018-08-23:
Hi Jack,

I need you to copy a reproducible example or at least show me the structure of your data. Otherwise I can't figure ou what caused the problem. Duplicated seed IDs means that you have two tree tops with the same ID thus you will have two trees that share the same ID.

Jack-Hastings, 2018-08-23:
Hi Jean-Romain --
To me it seems that the code is interpreting 'treetops' as a RasterLayer instead of a data.frame.
I used tree_detection() to generate 'treetops'
I hope this is enough info:

tree_top <- tree_detection(LAS, 3, 15)  
crowns_dalponte  <-  lastrees_dalponte(LAS, CHM, tree_top, extra = T)
Error: Duplicated seed IDs.

>LAS
class        : LAS (LASF v1.3)
memory       : 12.1 Mb 
extent       : 731591.8 , 731815.4 , 4713499 , 4713723 (xmin, xmax, ymin, ymax)
area         : 49991.69 m² (convex hull)
points       : 397039 points,  NA pulses
density      : 7.94 points/m²,  NA pulses/m²
field names  : X Y Z Intensity Classification 
coord. ref.  : +init=epsg:32618 +proj=utm +zone=18 +datum=WGS84 +units=m +no_defs +ellps=WGS84 +towgs84=0,0,0 

> str(tree_top)
Classes ‘data.table’ and 'data.frame':	2422 obs. of  3 variables:
 $ X: num  731733 731708 731693 731662 731682 ...
 $ Y: num  4713501 4713500 4713500 4713500 4713501 ...
 $ Z: num  22.5 27.1 29.6 19.9 21.4 ...
 - attr(*, ".internal.selfref")=<externalptr>

Jean-Romain, 2018-08-23:
Which version of the package are you using ? Looking at your data it seems to me that you encountered a fixed bug.

Jack-Hastings, 2018-08-23:
> packageDescription("lidR")
Package: lidR
Type: Package
Title: Airborne LiDAR Data Manipulation and Visualization for Forestry Applications
Version: 1.6.0
Date: 2018-07-25

Jean-Romain, 2018-08-23:
According to my logs it have been fixed in version 1.6.1 avaible on CRAN since yesterday. Lucky you :wink:

Jean-Romain, 2018-08-23:
Well actually it is not available for windows yet. Wait maybe one day more.

closed 2018-08-23 by Jean-Romain

Jack-Hastings, 2018-08-23:
Great! I'll sit tight through the weekend. Thank you for being so responsive and helpful.

Jack

Jean-Romain, 2018-08-23:
Optionally you can do:

```
ttops$Z = 1:nrow(ttops)
```

To by pass the bug.