r-lidar/lidR #168: lasclip outputs a list instead of a LAS

spono, 2018-08-23:
Ciao JR,
I installed yesterday the stable 1.6.1 release from CRAN and I found out that `lasclip` used with a `SpatialPolygonDataframe` outputs a list (with a LAS inside) and not anymore a LAS. Can you confirm this behaviour? is it meant to be like this or is it a regression?
thank you in advance

Jean-Romain, 2018-08-23:
`lidR 1.6.0` was released on 2018-07-26. The lastest revision of `lasclip` was  2018-07-19. So my answer is no it is not expect to get a different output. And yes clipping with a spdf is expected to return a list. Indeed you can have several polygons.

spono, 2018-08-23:
my fault!
I realised only now that in the Wiki is correctly explained:

> In the first case a single LAS object is returned and in the second case a list of LAS objects is returned

sorry

closed 2018-08-23 by Jean-Romain