r-lidar/lidR #169: [1.6.1] lastrees_li2: set distance limit to segmentation

spono, 2018-08-23:
Ciao JR,
using `lastrees_li2` I can see that the process includes in the segmented tree all the ground points within the provided maximum crown radius. While this works well for trees in a forest where few space is shared by many (and returns a close-to-real crown occupancy), if I segment an isolated individual, I get the tree with all the points within the `speedup` radius.

Would it be (reasonable and) feasible to set a sort of threshold below which the process includes only the points within the identified crown area or not farther then e.g. 2m from the tree axis?
all the best

Jean-Romain, 2018-08-23:
Ciao

My answer is no. I coded the algorithm as it is described in the original paper. This is a known limitation of the method and I will not change that. What I sometime change in my code compared to original publication are minor points that can be understood without a deep documentation (example `hmin` that stop the segmentation early). But I'm not going to change deeper stuff because if I change stuff it will not anymore be the methods described in the paper.

However what I can do is to create a new algorithm. And spoiler I will maybe do it. I'm currently comparing many methods and if the li et al. method appears to be good at tree detection I will use the main idea for the tree detection but I will modify the segmentation in a way that it faster less prone to obvious issues as you reported. But this is not for a close future.

closed 2018-08-23 by Jean-Romain

spono, 2018-08-23:
fair! ;)
I'll wait for future news, then. Keep up with the good work!