r-lidar/lidR #170: Extracting waveform from point cloud data

thatsciencegal, 2018-08-23:
Hi Jean-Romain,

I'm trying to extract a "waveform" from the point cloud data I processed (basically by creating a histogram of the number of returns in each canopy bin). I think these bins are made for the entropy function, but I didn't see anything that outputs n for the height bins chosen. Am I overlooking something or is there some other process I should use to get at this information?

Thanks!

-Christine

Jean-Romain, 2018-08-23:
I'm not sure to understand what do you mean. Are you talking about full waveform lidar ? FW data files are not supported.

If you want to create an histogram of return per bin this is only regular R code not related to `lidR`.

```r
hist(las@data$Z, breaks = seq(0, ceiling(max(las@data$Z)), 1))
```

There is also the function `findInterval` that can help.

closed 2018-08-24 by Jean-Romain

thatsciencegal, 2018-08-27:
Sorry, I didn't explain it well. I am using the gridmetrics function to subdivide my lidar data into different sizes (e.g., 5x5 m, 10x10 m, etc.). I want to get the histogram of returns per bin for each grid cell. Is there something within gridmetrics (or another function) where this information can be pulled? I imagine gridmetrics is doing this somewhere in the background because I think that's where the entropy calculation comes from.

Jean-Romain, 2018-08-27:
`grid_metrics` feeds a user-defined function with the point within each cells and aggregate the output as long as the user-defined function returns *numbers*. For example `entropy` returns a number.

Here what you want is not to return a number but a vector of numbers. This is not how `grid_metrics` is expected to works. However in version < 1.6.1 we can hack the function by using a `list`.

```r
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las = readLAS(LASfile)

myMetrics = function(z, by = 1)
{
  zmax = max(z)
  breaks = seq(0, ceiling(zmax/by)*by, by)
  h = hist(z, breaks = breaks)
  return(list(histogram = list(h)))
}

X = grid_metrics(las, myMetrics(Z, by = 1), 20)

plot(X$histogram[[1]])
```

But this trick will no be possible anymore in future versions because `grid_metrics` will return `Raster*` objects.