r-lidar/lidR #172: catalog_apply stops with rlas::check_data_vs_header error

pmagdon, 2018-09-12:
When using catalog_apply() to run processsing on a catalog of las 1.3 files I get the following error:
Error in rlas::check_data_vs_header(header, data, hard = FALSE) : 
  The header describes extra bytes attributes that are not in the data.

However, when I run the same processing on a single file out of the catalog using readLAS everything is fine. Is there an option to prevent the check or to fix the header?  Is there an option to check which extra byte is causing this problem?

Jean-Romain, 2018-09-12:
Ok I remember now. Install the lastest available version of `rlas` on CRAN which is `1.2.7`. This should solve your issue.

pmagdon, 2018-09-13:
Thanks for the quick repons. Yes update of rlas from Cran solves the issue!

closed 2018-09-13 by pmagdon