r-lidar/lidR #179: WriteLAS loses CRS when writing.

bw4sz, 2018-09-21:
I feel like this issue was mentioned somewhere, but I can't seem to find it in closed? 

May be related to an issue I submitted in the spring. https://github.com/Jean-Romain/lidR/issues/126

```
> clipped_las
class        : LAS (LASF v1.3)
memory       : 412.8 Kb
extent       : 255678.9 , 255718.9 , 4109807 , 4109847 (xmin, xmax, ymin, ymax)
area         : 1595.001 m² (convex hull)
points       : 5100 points,  NA pulses
density      : 3.2 points/m²,  NA pulses/m²
field names  : X Y Z gpstime Intensity ReturnNumber NumberOfReturns ScanDirectionFlag EdgeOfFlightline Classification Synthetic_flag Keypoint_flag Withheld_flag ScanAngle UserData PointSourceID
coord. ref.  : +init=epsg:32611 +proj=utm +zone=11 +datum=WGS84 +units=m +no_defs +ellps=WGS84 +towgs84=0,0,0*

> lidR::writeLAS(clipped_las,"test.laz")
> reload<-readLAS("test.laz")
> reload
class        : LAS (LASF v1.3)
memory       : 408.9 Kb
extent       : 255678.9 , 255718.9 , 4109807 , 4109847 (xmin, xmax, ymin, ymax)
area         : 1595.001 m² (convex hull)
points       : 5100 points,  NA pulses
density      : 3.2 points/m²,  NA pulses/m²
field names  : X Y Z gpstime Intensity ReturnNumber NumberOfReturns ScanDirectionFlag EdgeOfFlightline Classification Synthetic_flag Keypoint_flag Withheld_flag ScanAngle UserData PointSourceID
*coord. ref.  : NA*

> sessionInfo()
R version 3.4.3 (2017-11-30)
Platform: x86_64-pc-linux-gnu (64-bit)
Running under: Ubuntu 16.04.5 LTS

Matrix products: default
BLAS: /usr/lib/libblas/libblas.so.3.6.0
LAPACK: /usr/lib/lapack/liblapack.so.3.6.0

locale:
 [1] LC_CTYPE=en_US.UTF-8       LC_NUMERIC=C
 [3] LC_TIME=en_US.UTF-8        LC_COLLATE=en_US.UTF-8
 [5] LC_MONETARY=en_US.UTF-8    LC_MESSAGES=en_US.UTF-8
 [7] LC_PAPER=en_US.UTF-8       LC_NAME=C
 [9] LC_ADDRESS=C               LC_TELEPHONE=C
[11] LC_MEASUREMENT=en_US.UTF-8 LC_IDENTIFICATION=C

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base

other attached packages:
[1] lidR_1.6.1

loaded via a namespace (and not attached):
 [1] Rcpp_0.12.18               magrittr_1.5
 [3] knitr_1.20                 raster_2.6-7
 [5] units_0.6-0                assertive.data.us_0.0-1
 [7] spDataLarge_0.2.7.0        assertive.code_0.0-1
 [9] lattice_0.20-35            assertive.reflection_0.0-4
[11] stringr_1.3.1              tools_3.4.3
[13] assertive.matrices_0.0-1   assertive.properties_0.0-4
[15] rgdal_1.3-3                grid_3.4.3
[17] data.table_1.11.4          e1071_1.6-8
[19] DBI_1.0.0                  rgeos_0.3-28
[21] class_7.3-14               assertive.files_0.0-2
[23] rlas_1.2.4                 assertive.models_0.0-1
[25] digest_0.6.15              assertive.base_0.0-7
[27] sf_0.6-1                   spData_0.2.9.0
[29] assertive.data_0.0-1       settings_0.2.4
[31] assertive.datetimes_0.0-2  assertive.types_0.0-3
[33] codetools_0.2-15           assertive.sets_0.0-3
[35] memoise_1.1.0              glue_1.3.0
[37] sp_1.3-1                   stringi_1.2.4
[39] compiler_3.4.3             assertive.data.uk_0.0-1
[41] assertive.numbers_0.0-2    assertive.strings_0.0-3
[43] classInt_0.2-3             assertive_0.3-5
```

Great work, looking forward to 2.0!

Jean-Romain, 2018-09-21:
Yes this is a known issue. `rlas` the package that drives las I/O is not yet able to write the EPSG code. I will try to improve this point in `rlas` but it is not related to `lidR`

Jean-Romain, 2018-09-23:
`rlas 1.2.8` avaible on github https://github.com/Jean-Romain/rlas writes the epsg code into the las file. The information is not longer lost. I will now make what is needed in lidR to take advantage of that using the `proj4string` information

gisma, 2018-09-23:
Great! This would be awesome!

Jean-Romain, 2018-09-23:
In `lidR v2.0`.

```r
# Load a file with no CRS:
LASfile <- system.file("extdata", "example.laz", package="rlas")
las = readLAS(LASfile)

# Set the CRS via the function `epsg`. 
# To understand why `epsg` and not `projection` refer to the documentation:
epsg(las) <- 26917
las

# Updating rlas the EPSG code is written, the CRS is preserved
f = tempfile(fileext = ".las")
writeLAS(las, f)
readLAS(f)
```

In `lidR < 2.0` you can't modify the header easily with a built in function but the `rlas` write capabilities are still available by installing the latest version of `rlas`. You must modify the header manually.

```r
LASfile <- system.file("extdata", "example.laz", package="rlas")
las = readLAS(LASfile)

las@header@VLR$GeoKeyDirectoryTag$tags <- list()
las@header@VLR$GeoKeyDirectoryTag$tags[[1]] <- list(key = 3072L, `value offset` = 26917L)

f = tempfile(fileext = ".las")

writeLAS(las, f)
readLAS(f)
```

In `lidR < 2.0` if the epsg code was already recorded in the read file it is preserved by updating `rlas`

```r
# A las file with an epsg code recorded.
las = readLAS(LASfile)

f = tempfile(fileext = ".las")
writeLAS(las, f)
readLAS(f)
```

closed 2018-09-23 by Jean-Romain

Jean-Romain, 2018-10-27:
`rlas 1.2.9` available on CRAN

ameztegui, 2018-11-10:
Hi Jean-Romain,

I isntalled devel versions of both rlas and lidR, an when I try to apply the function epsg(las) as provided in the example, I get the following error
```
Error in (function (cl, name, valueClass)  : 
  ‘crs’ is not a slot in class “LAS”
```
And indeed, `las` object doesn't seem to contain a slot called `crs` anymore. How should I proceed then?

Jean-Romain, 2018-11-10:
In lidR v2 LAS objects were redesigned and inherit a `Spatial` object from `sp`. Thus there is no longer any slot `crs`. It is replaced by `proj4string` like other object from `sp`.

In your case you did not provide enough code to help me to figure out what is going wrong. The fact that the words `slot` and `crs` appear in the same sentence make me think that R is confused between old and new version. I suggest to reinstall and/or give me more context. The example code works on my computer.

ameztegui, 2018-11-14:
Reinstalling both packages worked. You were right, somehow R was kind of confused about package versions. 
Merci beaucoup Jean-Romain!