r-lidar/lidR #180: After calling "plot" nothing happens...

gponce-ars, 2018-09-25:
Hi, 
I'm following the example from [https://github.com/Jean-Romain/lidR/wiki/lasclassify](url). However, when I get to the first `plot` the image never shows up,  no errors,  just doesn't display anything.  If I do `plot(cars)` the plot shows up fine. 

The `lidR` installation went smoothly with no errors on macOS High Sierra 10.13.6.  I just did the `install.packages("lidR")`.

The example trying to run:
```
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
shapefile_dir <- system.file("extdata", package = "lidR")
    
las = readLAS(LASfile)
lake  = rgdal::readOGR(shapefile_dir, "lake_polygons_UTM17")
lasclassify(las, lake, field="lake")
forest = lasfilter(las, lake == FALSE)
# **** After running these plot, nothing happens. ****
plot(las)
plot(forest)  

```
Here is my session Info.

```
> sessionInfo()
R version 3.5.0 (2018-04-23)
Platform: x86_64-apple-darwin15.6.0 (64-bit)
Running under: macOS High Sierra 10.13.6
Matrix products: default
BLAS: /Library/Frameworks/R.framework/Versions/3.5/Resources/lib/libRblas.0.dylib
LAPACK: /Library/Frameworks/R.framework/Versions/3.5/Resources/lib/libRlapack.dylib
locale:
[1] en_US.UTF-8/en_US.UTF-8/en_US.UTF-8/C/en_US.UTF-8/en_US.UTF-8
attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base
other attached packages:
[1] lidR_1.6.1
loaded via a namespace (and not attached):
 [1] Rcpp_0.12.18               assertive.data_0.0-1
 [3] compiler_3.5.0             later_0.7.3
...

```
Every time I do something like `plot(las)` or `plot(forest)` I start to see these RGL devices on my X11 window (Xquartz) as you can see in the image below, however, nothing gets displayed, and already switched between the different X11  devices...

[image]

Any hint?

gponce-ars, 2018-09-26:
Thank you.
Something might be wrong with my current rgl - X11 (quartz) interaction.

I just tested in RedHat 7.5 and everything works fine.

Do you have documentation on using PointCloudViewer? I'm trying to use `plot_xyzrgb` with some data I read with `lidR`. 

G

closed 2018-09-27 by Jean-Romain

ENIAC-6, 2022-11-16:
This happened to me too. I think it would be helpful to explicitly mention in 2.3.1 Basic 3D rendering:

library(rgl)
plot(las)

ENIAC-6, 2022-11-16:
I see, rgl plot does not open in Plots window in RStudio. Yes, runs without library(rgl) on Windows, just opening in new dialog box.
 
Thank you!

Jean-Romain, 2022-11-16:
Are we talking about Mac or Windows. This issue was about Mac

ENIAC-6, 2022-11-16:
Got it, this was Windows

Sillson, 2023-04-26:
For anyone else who comes across this issue...I'm on a 2018 Mac and was able to overcome this by running : `rglwidget()` directly before each plot.