r-lidar/lidR #181: Determining area in a point cloud using spefice heights

Selenachav, 2018-09-26:
Hello,
iI there a way to determine the area of a point cloud based on a specific range in the points clouds height. In essence, I am trying to determine the area of my canopy based on the estimated height range I calculated.

I assume I can use both the filter and area function but don't know how I could filter my point cloud to a specific height range such as 16m to 13m

Jean-Romain, 2018-09-27:
`lasarea` does not exists anymore (since a long time). You read an outdated documentation. Please tell me were you read that, I will fix it.

I don't understand your main question. The area of the point cloud within the layer 13-16 does not have a meaning. Actually the area of a point cloud is meaningless too. But with dense and regularly distributed point clouds one may easily understand the meaning of  'area' as 'covered area'. Moreover it is documented, it is approximated by the convex hull area of the point cloud. In your case I want to say that the covered area in a given layer cannot be computed with a convex hull because points are not regularly distributed.

bi0m3trics, 2018-09-27:
I offer this approach as an alternative too, if I understood you correctly, which approximates the area with at least one point at a resolution of 0.5m…
```
# Apply your filter
randomFilter = lasfilter(lidar, Z > 13 & Z <= 16)

# create a raster of counts per unit area (res) for your desired minimum mapping unit
d = grid_metrics(randomFilter, length(Z), res=0.5)
d = as.raster(d)

# then multiply the sum by the squared cell size to get the “area occupied" by points
sum(d[] >= 1, na.rm = TRUE) * raster::res(d)[1]^2
```
It may go without saying, but this is different from JR’s convex hull “area occupied” interpretation.
-bi0m3trics

Selenachav, 2018-09-27:
I found the documentation for the lasarea below but I did just notice that it is the documentation for an older version of the package.
https://www.rdocumentation.org/packages/lidR/versions/1.0.2/topics/lasarea

closed 2018-09-27 by Selenachav