r-lidar/lidR #182: grid_canopy error

muzo583, 2018-10-11:
Hi,
I have get an error.

```
LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las <- readLAS(LASfile, select = "xyz", filter = "-drop_z_below 0")
col <- pastel.colors(200)

chm <- grid_canopy(las, 0.5, p2r(0.3))
ker <- matrix(1,3,3)
chm <- raster::focal(chm, w = ker, fun = mean, na.rm = TRUE)

ttops <- tree_detection(chm, lmf(4, 2))
las   <- lastrees(las, dalponte2016(chm, ttops))
plot(las, color = "treeID", colorPalette = col)
```
It is an example code.
An error is

Error in C_grid_canopy(las, raster::as.matrix(bbox), raster::res(layout)[1],  : 
  Expecting a single value: [extent=4].
Any idea? 

thanks

Jean-Romain, 2018-10-11:
This code works on my computer. Did you strictly copy/paste the code that raise this error? Did it occur with another dataset that the one provided as example?

Jean-Romain, 2018-10-11:
Ok so try to reinstall. Basically your error states that the C++ function is badly called but it should not since the internal code is correct. This issue is maybe related to that one https://github.com/Jean-Romain/lidR/issues/178 which is Windows specific and not reproducible...

Do other Windows user can reproduce? @bi0m3trics?

muzo583, 2018-10-11:
I have converted lidr v 1.6. It works. I will try reinstall as your commented. Thanks

bi0m3trics, 2018-10-11:
I just copied and pasted his code form the original post and it works for me running 2.0.0 on windows 10 in R 3.5.0.