r-lidar/lidR #183: multicore grid_metrics

samherniman, 2018-10-11:
Hey JR

I'm using lidR 2.0.0
When I try to run grid_metrics with multiple cores, I get an error.
When I run it with one core, it works perfectly.
Relevant code is below:

    ctg = catalog("D:\\Newfoundland\\Data\\ALS\\HR\\normalized\\20180730")
    opt_cores(ctg) <- 8L
    
    myMetrics = function(z)
    {
      metrics = list(
        z20 = quantile(z, .2),
        z80 = quantile(z, .8),
        zsd = sd(z)
      )
      
      return(metrics)
    }

    hrhigh = grid_metrics(ctg, myMetrics(Z), 15)

The error looks like this:

> Error in myMetrics(Z) : could not find function "myMetrics"

Again, it works when I change the number of cores to 1. 

Thanks!
Sam

Jean-Romain, 2018-10-11:
Ok that one is a known one and is MS Windows specific (that's why I missed it). I already fixed it in previous version but reintroduced in this new version when simplifying the underlying code. I'll fix that. Thank you for reporting the issue.

bi0m3trics, 2018-10-11:
I ran into this too with something I was just testing in 2.0.0 on windows and assumed it was a design feature ;) 
In the interim, I think you could just replace your call at the end to `grid_metrics` with a call using catalog_apply
e.g., 
```
myMetrics_Catalog = function(cluster, res)
{
  las = readLAS(cluster)
  if (is.empty(las)) return(NULL)
 
  metrics <- grid_metrics(las, myMetrics(Z), res)
  bbox   <- raster::extent(cluster)
  metrics <- raster::crop(metrics, bbox)
  
  return(metrics)
}
hrhigh = catalog_apply(ctg, myMetrics_Catalog, res = 15)
hrhigh = do.call(raster::merge, hrhigh )
```

Jean-Romain, 2018-10-11:
@bi0m3trics you're right I guess it works. It means that I actually fixed this issue in the generic catalog processing engine but in `grid_metrics` the user's functions  are  still not exported in each clusters. This information may help me.

Jean-Romain, 2018-10-11:
That one was a nasty one and implied to rewrite a lot of code! I broke a lot of stuff internally but everything is now fixed according to my batch of automatic tests. I hope I didn't break more stuff.

Several points here:

1. @samherniman's code works
2. This code should be better written like that for computation speed (avoid to sort twice the Z vector)

```r
myMetrics = function(z)
{
  props = c(0.2, 0.8)
  zq = as.list(stats::quantile(z, props))
  names(zq) = c("z20", "z80")
  metrics = c(zq, zsd = sd(z))
  return(metrics)
}
```

3. @bi0m3trics code worked because `myMetrics` is explicitly called within a non lazyly evaluated function and thus is findable by automatic code inspection. But this option have two drawbacks: (1) `do.call(raster::merge, ...)` loose the metric names, (2) strict continuity of the output raster is not guarantee because the chunks or the tiles are not necessarily aligned with the raster depending of the chosen starting points and resolution. What should be done is the following:

```r
myMetrics_Catalog = function(cluster, res)
{
  las = readLAS(cluster)
  if (is.empty(las)) return(NULL)
 
  metrics <- grid_metrics(las, myMetrics(Z), res)
  bbox   <- raster::extent(cluster)
  metrics <- raster::crop(metrics, bbox)
  
  return(metrics)
}

hrhigh = catalog_apply(ctg, myMetrics_Catalog, res = 15, check_alignment = TRUE)

names <- names(hrhigh[[1]])
hrhigh <- do.call(raster::merge, hrhigh)
names(hrhigh) <- names
if (is(hrhigh, "RasterBrick")) colnames(hrhigh@data@values) <- names
```

But the option `check_alignment` is not yet available to the user.

samherniman, 2018-10-12:
Thanks guys!

commit 18a0ba8d (2018-10-26): Fix #183
commit 1e93f9c2 (2018-12-03): v2.0.0 (#195)

* lidR v2.0.0  (#176)

* lidR v2.0.0 with more than 200 commits

* Fix progress bar bug

* Fix progress bar

* rewrite plot.LAS and remove lascolor

* Fix INT16_MIN missing + -Wsign-compare + -Wreorder

* Fix INT32_MIN' was not declared in this scope

* Try to fix linking to openblas on windows

* #179 support of EPSG code and CRS

* #177 No longer any update by reference of LAS objecta.

* Remove C++ function to update header by ref

* New function lastransform

* New plot method for LASheaders

* Remove lastest trace of lasmetrics class

* add final blank lines

* Document + t [... cut]
closed 2018-10-12 by Jean-Romain