r-lidar/lidR #184: lasnormalize bug on lidR 2.0

tiagodc, 2018-10-15:
Hi there, I've been trying to run `lasnormalize` with the DTM as input for ground reference and I've been getting the following error: 

`Error: 1 points were not normalizable because the DTM contained NA values. Process aborded.`

It happens even if I remove or replace the NAs in the DTM raster object - which was generated using the `grid_terrain` function. I only got it to work when setting the DTM pixel resolution to 5m or more, for higher resolutions the error happens recurrently.

Hope it has an easy fix!

Cheers

Jean-Romain, 2018-10-15:
Were does your DTM come from ? If it comes from an external source it is not a bug but a feature. If the DTM comes from `grid_terrain` I must fix that (again).

What happened here is that 1 points was not associated with a pixel of the DTM. This point is likely to be on the very edge of the DTM.

1. If the DTM come from an external file it means that your DTM does not encompass your data (error of a single point). 
2. If the DTM comes from `grid_terrain` the problem comes from the fact that the ground points don't encompass all the other points and thus the DTM can't encompass all the points. To fix that I added extra interpolation within a additional buffer but maybe in your case this buffer is too tiny because your data have a weird shape or a locally low point density or any other data specific issue I was not able to predict.

tiagodc, 2018-10-15:
The DTM was generated using `grid_terrain` - all data processing was done in lidR actually.

The point cloud I'm using is from terrestrial laser scanning, so the point density is very high around the scanner, and gets progressively lower at the edges. Here follows an image:

[image]

tiagodc, 2018-10-15:
OK, that should solve it. Thanks!