r-lidar/lidR #187: lidR::plot3d gives "unable to find an inherited method for function..." error

Erikpinter, 2018-10-25:
Hej Jean-Romain,

first of all: **thank you for your wonderful work**! I've been using your package for the detection of trees with Mobile Rail LiDAR data. 

Unfortunately recently I've failed to run some of my code, old where I use your plot3d function. 
example:
```
require(lidR)

LASfile <- system.file("extdata", "Topography.laz", package="lidR")
lidar = readLAS(LASfile)
shapefile_dir <- system.file("extdata", package = "lidR")

dtm =   grid_terrain(lidar, res = 1, method = "knnidw", k = 6L)
lidR::plot3d(dtm)

```

It keeps throwing the following error:

> Error in (function (classes, fdef, mtable)  : 
>   unable to find an inherited method for function ‘yFromRow’ for signature ‘"RasterLayer", "missing"’

Converting the dtm to a RasterLayer throws the same error:
```
rdtm <- as.raster(dtm)
lidR::plot3d(dtm)
```

Any help would be appreciated,

  Regards, Erik

My SessionInfo:

> R version 3.5.0 (2018-04-23)
> Platform: x86_64-w64-mingw32/x64 (64-bit)
> Running under: Windows >= 8 x64 (build 9200)
> 
> Matrix products: default
> 
> locale:
> [1] LC_COLLATE=German_Austria.1252  LC_CTYPE=German_Austria.1252   
> [3] LC_MONETARY=German_Austria.1252 LC_NUMERIC=C                   
> [5] LC_TIME=German_Austria.1252    
> 
> attached base packages:
> [1] stats     graphics  grDevices utils     datasets  methods   base     
> 
> other attached packages:
> [1] lidR_1.6.1
> 
> loaded via a namespace (and not attached):
>  [1] Rcpp_0.12.19               knitr_1.20                
>  [3] raster_2.7-15              assertive.data.us_0.0-2   
>  [5] assertive.code_0.0-3       lattice_0.20-35           
>  [7] assertive.reflection_0.0-4 tools_3.5.0               
>  [9] assertive.matrices_0.0-1   assertive.properties_0.0-4
> [11] grid_3.5.0                 data.table_1.11.8         
> [13] rgeos_0.4-1                rlas_1.2.7                
> [15] assertive.files_0.0-2      assertive.models_0.0-2    
> [17] digest_0.6.18              yaml_2.2.0                
> [19] assertive.base_0.0-7       assertive.data_0.0-1      
> [21] settings_0.2.4             assertive.datetimes_0.0-2 
> [23] assertive.types_0.0-3      codetools_0.2-15          
> [25] assertive.sets_0.0-3       memoise_1.1.0             
> [27] glue_1.3.0                 sp_1.3-1                  
> [29] stringi_1.2.4              compiler_3.5.0            
> [31] assertive.data.uk_0.0-2    assertive.numbers_0.0-2   
> [33] assertive.strings_0.0-3    assertive_0.3-5

Jean-Romain, 2018-10-25:
Hi, your example works on my machine. Do you have more clues? Did it use to work? If yes when? on which R version it worked the last time? I'm still running under R 3.4.4.

Jean-Romain, 2018-10-25:
Ok I reproduced the bug by updating the `raster` package. The issue comes from `raster`. I will fix that in version 2.0.0

Jean-Romain, 2018-10-25:
I think that the issue should be reported to `raster`. The following code does not work:

```r
library(raster)
r = raster()
res(r) <- 1

xFromCol(r)
yFromRow(r)
```

I can fix it like that

```r
yFromRow(r, 1:nrow(r))
xFromCol(r, 1:ncol(r))
```

But not before the new major release. Below the code of the fixe `plot3d` if you really need it.

```r
plot3d = function(x, y, add = FALSE, bg = "black", ...)
{
  inargs <- list(...)

  if(!is(x, "RasterLayer"))
    stop("Object not supported")

  mx =  t(apply(raster::as.matrix(x), 2, rev))
  x_  = sort(raster::xFromCol(x, 1:raster::ncol(x)))
  y_  = sort(raster::yFromRow(x, 1:raster::nrow(x)))

  if(!add)
    rgl::open3d()

  rgl::rgl.bg(color = bg)
  rgl::surface3d(x_, y_, mx, front="lines", col="white", ...)
}
```

closed 2018-10-25 by Jean-Romain

Erikpinter, 2018-11-03:
Thanks a lot! The workaround works perfectly!