r-lidar/lidR #196: Wiki taken down

EnvSci, 2018-11-30:
Dear Jean-Romain, Wish you had kept the wiki until the V2 is available. the wiki was helping me a lot. Now suddenly its out.
Thanks
Syed

Jean-Romain, 2018-11-30:
I understand your concern. In the meantime there are several reasons why I removed the wiki pages:

1. lidR have more an more users. When I will release the version 2.0 of the package that is deeply incompatible with the previous releases I will be overwhelmed by user complaining. Anyway I had to re-engineer the package. Removing the wiki I force some users to be aware of some important changes that are coming soon. And look ! It works :wink: 
2. I cannot maintain the wiki pages any more. It takes a lot of time, some of them are outdated, some of them are useless, some of them should be re-organised. And each of them must be updated for the new major release. Instead I'm currently making some vignettes that will replace some wiki pages. Vignette are semi-automatically generated and thus implies less work.

It was ok to make some wiki pages at the very beginning. Now the lidR packages takes me more and more time. I have to take decisions to spend less time on the package. If you want to get the document back you can contribute by writing vignettes for the new release.

closed 2018-12-19 by Jean-Romain

Jean-Romain, 2019-01-02:
Wiki pages are back (at least some of them).

EnvSci, 2019-01-03:
Thank you for getting Wiki pages back online. :)