r-lidar/lidR #200: High similarity in results using the different algorithms for DSM DTM generation

gisma, 2018-12-19:
Hi Jean-Romain 

I recently was checking the different results of generation DSM,DTM and CHM. Actually I am hardly able to construct any differences with the DTM generation and statistically spoken almost no differences while using the various DSM algorithms. I am just wondering if I make a bad mistake or the results are really similar. if you have time I would be glad if you could take a look at the code example on the [github wiki page](https://github.com/gisma/uavRst/wiki/Canopy-Height-Model-from-Lidar-data-using-lidR).

cheers chris

Jean-Romain, 2018-12-19:
I updated and re-uploaded this wiki page: [Rasterizing perfect canopy height models](https://github.com/Jean-Romain/lidR/wiki/Rasterizing-perfect-canopy-height-models).

You can see (and test yourself) that the algorithms are actually different. However how different they are may depend on your dataset. In the example given in wiki page the `pitfree` algorithm performs very well. I tested the `pitfree` in an eucalyptus plantation 30 pt/m². Besides the fact it was obviously slow to compute, the output was not very nice. A simple point-to-raster method gave me better results.

That being said, your question was _"is it an expected behavior to get very close results"_ ? Well, I don't know. I implemented the algorithms but I did not test them into a wide variety of contexts. I guess that in some context and with some parameters the algorithms can provide very similar results. 

In you specific case and looking only at the `pitfree - tin` the two CHM are absolutely equal. That may be expected if your parameters are not adequate. After all the `pitfree` is nothing else than a serie of `tin` with height and triangle size filters (read the paper if not done yet). Here, I guess that in `pitfree(c(0,2,5,10,15), c(0, 1.5))` the value `1.5` is not good for your dataset. As a consequence no triangle are removed and consequently the `pitfree` is strictly equal to `dsmtin` (with useless extra computations) .

gisma, 2018-12-19:
Jean-Romain,
thanks for the comprehensive answer. 
I should have mentioned that with the `pitfree` and ` tin` algorithm I have been fine because I understood it as you explained it. 
Nevertheless my basic question was  not really _"is it an expected behavior to get very close results"_ It was more something like _"did I wrong or are the alogrithm really that robust"_. 
And I have to admit and should know that you are dealing with the first half of the question and I have to deal with the second by myself.  As not mentioned  there are  identical results with the `sf`and `pmf` approaches for classifying ground points. Basically I did recently  quite a bit of sensitivity studies for tree segmentation and tree finding algorithms and was a bit surprised that the algorithm are tending to  converg significantly. I will give your points a deeper glimpse.

So thanks for clarifying and very very big thanks for the great package!

Jean-Romain, 2018-12-19:
You're welcome. I'm glad to see that you already make an intensive usage of the LAScatalog processing engine :+1:

closed 2018-12-19 by Jean-Romain