r-lidar/lidR #201: LAS 1.4

jfprieur, 2018-12-19:
Hi Jean-Romain,

There is not much to be done (I think it is a liblas problem) but I thought you should be aware that i ran into some strange errors when I use LAS 1.4 formatted files. I tried doing the tiling process with this new data and one tile would be written in the directory, then it would be erased, then the new tile, then erased, etc.

When I converted the LAS to 1.2 format using las2las everything worked fine as usual. This is on lidR v2.

As  I said, I have always had problems with 1.4 format files, I know in Python there is a library called laspy that addresses liblas shortcomings, do not know if there is such a package in R that we could try.

Just thought you should be aware(!) if you were not already in case you get questions. Thanks again for the great R package!

Jean-Romain, 2018-12-19:
Hi,

This bug is likely to come from the `rlas` package. `lidR` writes sequentially the retiled files but files with 0 points are automatically removed. The point is that I have never seen a LAS 1.4 file. I don't even know if it is supported properly. According to this issue I guess that `rlas` writes 0 points. Please give me a small (e.g. 400 by 400 m) LAS 1.4 file and I will study this case.

Jean-Romain, 2018-12-19:
Well, after thinking more about it, if LAS 1.4 files are read (no even sure of that) this format cannot be fully supported yet. For example in the LAS 1.4 format the scan angle rank can be a decimal value. If `rlas` read the file it will read it as an integer anyway. Just an example.

Jean-Romain, 2018-12-19:
I'm pretty sure LASlib supports LAS 1.4. Actually I'm waiting for somebody complaining about LAS 1.4 and who can provide me a chunk of LAS 1.4 to work on the topic. I could be you :+1:

jfprieur, 2018-12-19:
Alright I will prepare a small tile for you and send it to you by email ;)