r-lidar/lidR #207: grid_terrain with a max_edge restriction for TIN interpolation

dncosenza, 2019-01-07:
I need to generate a DTM by running the 'grid_terrain' using the 'TIN' algorithm, but using the same restriction imposed by the 'max_edge' of 'grid_canopy(algorithm= dsmtin)'. I tried to execute the 'grid_canopy', but according to the pdf, it triangulates just the first return, so it won't work for DTMs (that is desirable to use lasts returns).
A good workaround could be to delete the first returns of the las data and then reclassify the last returns as first ones, and then run the  'grid_canopy(algorithm= dsmtin)' with 'max_edge' restriction. 
In any case, it is posted here the suggestion for future updates of 'grid_terrain' function.
Congratulations for your great job.

Jean-Romain, 2019-01-07:
This is a good suggestion however it is more difficult to implement that it may appears. Two points to consider:

1. The `tin` algorithm enable to interpolate the terrain withing the convex hull of the ground point *only*. That means that the DTM does not cover the full point cloud. Indeed the ground points are less numerous that other point and cover a narrower area. This is with the `tin` also performs a `knnidw` to fill the pixels that were not interpolated by a TIN.
2. If I enable the `max_edge` in `tin` the `knnidw` will fill the missing pixels anyway.

As a conclusion it is not an easy problem. The good solution is to work with a buffer. Load a wider region of interest than needed. Apply a modified version of `tin`. Clip the actual region of interest in the point cloud. Clip the DTM.

Here a modified version of `tin` that have a `max_edge` parameter and that does not perform a `knnidw`. As you can see the DTM does not cover the entire point cloud.

```r
tin2 = function(max_edge = 0) {
  f = function(what, where)  {
    context <- tryCatch({get("lidR.context", envir = parent.frame())}, error = function(e) {return(NULL)})
    lidR:::stopif_wrong_context(context, c("grid_terrain", "spatial_interpolation"), "tin2")
    z <- lidR:::interpolate_delaunay(what, where, max_edge)
    return(z)
  }  
  class(f) <- c("function", "SpatialInterpolation", "Algorithm", "lidR")
  return(f)
}

LAS = system.file("extdata", "Topography.laz", package="lidR")
las = readLAS(LAS)
x   = c(481340, 481340, 481280, 481300, 481280, 481340) - 148940
y   = c(3812940, 3813000, 3813000, 3812960, 3812940, 3812940) + 1425560
las = lasclipPolygon(las,x,y)
plot(las)

dtm = grid_terrain(las2, algorithm = tin2(15))

plot(dtm, col = terrain.colors(50))
```

dncosenza, 2019-01-08:
But instead to modify the 'grid_terrain', why not allow the 'grid_canopy' to use other classes of points (or returns) and let the user define which value (the highest, medium, etc...) that will be converted into a pixel value. If this is not a bad idea, it could be possible? Because It seems to me it will produce the same results and the DTM would be produced only inside the convex hulls under some 'max_edge' restriction.

Jean-Romain, 2019-01-08:
> Why not allow the `grid_canopy` to use other classes of points (or returns)

Because the algorithms you can use in `grid_canopy` guarantee to return a valid CHM according to the state of the art. If I allows users to apply `grid_canopy` algorithm with ground returns it does not make sense anymore. I won't do that.

> and let the user define which value (the highest, medium, etc...) that will be converted into a pixel value

This is basically the suggestion #205. If you want to contribute in #205  your opinions and your ideas will be appreciated.

> Because It seems to me it will produce the same results and the DTM would be produced only inside the convex hulls under some `max_edge` restriction.

The algorithms used in `grid_terrain` guarantee that every single point of the original point cloud will fall in a non empty pixel of the DTM. This way the output of `grid_terrain` can always be used in `lasnormalize`. What you are describing does have this property. What you are describing is exactly what I gave you yesterday and is bad.

So what you actually want is a function `grid_triangulation` that triangulate, interpolate and rasterize the triangulation of any set of point. It is a good idea and I could add such function in the next release with, in addition, a support of the suggestion #205

```r
grid_triangulation = function(las, res, max_edge = 0) 
{
  layout <- lidR:::make_overlay_raster(las, res)
  
  names(layout) <- "Z"
  grid <- raster::as.data.frame(layout, xy = TRUE)
  data.table::setDT(grid)
  grid[, Z := NULL]
  data.table::setnames(grid, names(grid), c("X", "Y"))
  
  points <- lidR:::coordinates3D(las)
  z      <- lidR:::interpolate_delaunay(points, grid, max_edge)
  cells  <- raster::cellFromXY(layout, grid)
  suppressWarnings(layout[cells] <- z)
  
  return(layout)
}

library(lidR)

LAS = system.file("extdata", "Topography.laz", package="lidR")
las = readLAS(LAS)
x   = c(481340, 481340, 481280, 481300, 481280, 481340) - 148540
y   = c(3812940, 3813000, 3813000, 3812960, 3812940, 3812940) + 1425560
las = lasclipPolygon(las,x,y)
plot(las)

ground = lasfilterground(las)
r = grid_triangulation(ground, 1, 15)
plot(r, col = terrain.colors(50))

second = lasfilter(las, ReturnNumber == 2)
r = grid_triangulation(second, 2, 15)
plot(r, col = terrain.colors(50))
```

closed 2019-02-21 by Jean-Romain