r-lidar/lidR #211: Issue with lasnormalize and grid_terrain

amithaldar87, 2019-01-18:
`lasc03_norm <- lasnormalize(lasC03, tin())`
Error in UseMethod("lasnormalize", las) : 
  no applicable method for 'lasnormalize' applied to an object of class "c('data.table', 'data.frame')"
 
`dtmc03 <- grid_terrain(lasC03, 0.05, kriging(k=10L))`
Error in UseMethod("grid_canopy", las) : 
  no applicable method for 'grid_canopy' applied to an object of class "c('data.table', 'data.frame')"

bi0m3trics, 2019-01-18:
You wouldn't happen to be a windows user would you? I tend to get these types of message in windows when packages don't load or load partially (it happens). I wonder if explicitly loading the `data.table` package would resolve the issue? If so, I typically reinstall the broken package (with `dep=TRUE`) or explicitly load faulty package when I see the "`no applicable method`" error, as that usually means R doesn't know what to do with the object it's being passed... and of course it should know.

Jean-Romain, 2019-01-18:
There is no mystery here, this is just the regular R behavior. You try to feed the function with the wrong object you received a regular R error. The functions expect a `LAS` object no a `data.frame` that's it. If you want to get started with lidR start with the example given in the documentation. Then if you have questions please ask on https://gis.stackexchange.com/. Thanks

closed 2019-01-18 by Jean-Romain