r-lidar/lidR #212: Issue with cataloge process with grid_metrics

komazsofi, 2019-01-18:
Dear Jean-Romain,

Thank you for the wonderful package! I am struggling with an error message during processing my data with LiDR. I am calculating area based LiDAR metrics for part of the Netherlands (for 4 tiles)  using LAScataloge functionality. The problem is that at the end of the calculation of LiDAR metrics sometimes gives me back the following error message, which I do not really understand where it comes from and how to fix it. (The code is working if I am only working with one tile but if I am building my catalog over 4 tiles it crashes if I try to calculate more than 2 features - VegStr_VertDistr_Metrics function in my code). I would like to ask whether you have some guess where the error comes from. 

 Processing [=====================================================================================================================================================================================] 100% (25/25) eta:  0s
Error in dimnames(x) <- dn : 
  length of 'dimnames' [2] not equal to array extent

Thank you in advance for your help,
Zsofia

My code is the following (full: build cataloge, derive DTM, calculate metrics): 

"
@author: Zsofia Koma, UvA
Aim: Pre-process AHN2 data
"
library("lidR")

# Functions
CoverageMetrics = function(z,classification)
{
  coveragemetrics = list(
    pulsepenrat = length(z[classification==2])/length(z),
    nofret_abovemean = length(z[z>mean(z)])/length(z)
  )
  return(coveragemetrics)
}

VegStr_VertDistr_Metrics = function(z)
{
  library("e1071")
  vertdistr_metrics = list(
    zstd = sd(z),
    zvar = var(z),
    zskew = skewness(z),
    zkurto = kurtosis(z),
    zentropy1=VCI(z,by=0.5,zmax=NULL),
    zentropy2=entropy(z+500, by = 0.5,zmax=NULL)
  )
  return(vertdistr_metrics)
}

# Set working dirctory
workingdirectory="D:/Koma/Paper1/ALS/wholestudyarea/"
setwd(workingdirectory)

resolution=2.5

# Create catalog
ctg <- catalog(workingdirectory)

opt_chunk_buffer(ctg) <- 5
opt_chunk_size(ctg) <- 1000
opt_cores(ctg) <- 7
opt_output_files(ctg) <- paste(workingdirectory,"tiled/{XLEFT}_{YBOTTOM}",sep="")

# Retile catalog
newctg = catalog_retile(ctg)

opt_chunk_buffer(newctg) <- 5
opt_cores(newctg) <- 7
opt_output_files(newctg) <- paste(workingdirectory,"ground/{XLEFT}_{YBOTTOM}_ground",sep="")

# Extract ground points
ground_ctg <- lasground(newctg, csf(sloop_smooth = TRUE))

# Create DTM
setwd(paste(workingdirectory,"ground/",sep=""))

pdf("LiDAR_process.pdf")

ground_ctg <- catalog(paste(workingdirectory,"ground/",sep=""))

ground_ctg@input_options$filter <- "-keep_class 2"
opt_chunk_buffer(ground_ctg) <- 5
opt_cores(ground_ctg) <- 18

dtm = grid_metrics(ground_ctg,min(Z),res=resolution)
crs(dtm) <- "+proj=sterea +lat_0=52.15616055555555 +lon_0=5.38763888888889 +k=0.9999079 +x_0=155000 +y_0=463000 +ellps=bessel +units=m +no_defs"
writeRaster(dtm, "dtm.tif",overwrite=TRUE)

plot(dtm)

# Calculate metrics into separate files per feature groups and classes -- point cloud based

coveragemetrics = grid_metrics(ground_ctg, CoverageMetrics(Z,Classification),res=resolution)
plot(coveragemetrics)

writeRaster(coveragemetrics,"coveragemetrics.grd",overwrite=TRUE)

vertdistr_metrics = grid_metrics(ground_ctg, VegStr_VertDistr_Metrics(Z),res=resolution)
plot(vertdistr_metrics)

writeRaster(vertdistr_metrics,"vertdistr_metrics.grd",overwrite=TRUE)

The datasets can be downloaded from here:

http://geodata.nationaalgeoregister.nl/ahn2/extract/ahn2_gefilterd/g02gz2.laz.zip
http://geodata.nationaalgeoregister.nl/ahn2/extract/ahn2_uitgefilterd/u02gz2.laz.zip

http://geodata.nationaalgeoregister.nl/ahn2/extract/ahn2_gefilterd/g02hz1.laz.zip
http://geodata.nationaalgeoregister.nl/ahn2/extract/ahn2_uitgefilterd/u02hz1.laz.zip

http://geodata.nationaalgeoregister.nl/ahn2/extract/ahn2_gefilterd/g06en2.laz.zip
http://geodata.nationaalgeoregister.nl/ahn2/extract/ahn2_uitgefilterd/u06en2.laz.zip

http://geodata.nationaalgeoregister.nl/ahn2/extract/ahn2_gefilterd/g06fn1.laz.zip
http://geodata.nationaalgeoregister.nl/ahn2/extract/ahn2_uitgefilterd/u06fn1.laz.zip

Jean-Romain, 2019-01-18:
Please edit your issue to make a **minimal** and reproducible example. Sadly I don't have time and I don't want to download and extract manually 8 zip files and read 80 lines of code in which I don't even know where it is expected to fail. Your code contains a lot of statement unrelated to the problem itself (why do you retile your dataset?).

The following works (I remove some metrics to do not have to install `e1071`). Furthermore you misused `VCI` (but I acknowledge that it is not well documented).

```r
library(lidR)

CoverageMetrics = function(z,classification) {
  coveragemetrics = list(
    pulsepenrat = length(z[classification==2])/length(z),
    nofret_abovemean = length(z[z>mean(z)])/length(z)
  )
  return(coveragemetrics)
}

VegStr_VertDistr_Metrics = function(z) {
  vertdistr_metrics = list(
    zstd = sd(z),
    zvar = var(z),
    zentropy1=entropy(z, by=0.5))
  
  return(vertdistr_metrics)
}

file <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = catalog(file)
opt_chunk_size(ctg) <- 160

metrics = grid_metrics(ctg, c(CoverageMetrics(Z,Classification), VegStr_VertDistr_Metrics(Z)), res = 10)
plot(metrics)
```

komazsofi, 2019-01-18:
Hi,

Sorry for the not appropriate error report (I will pay attention to minimizing my report related to the error next time). 

I attached the files because for me the error only occurred when I retiled this 4 big tiles and then I started to work with it (otherwise when I used one file as input then retile everything is worked fine). These files are big you cannot pre-process without retiling beforehand. 

Currently, I am re-running the code for my study side re-using your code. I will go back to you whether the error is still there.

Jean-Romain, 2019-01-18:
>  These files are big you cannot pre-process without retiling beforehand.

This is why there is the chunk size option in `lidR` accessible with `opt_chunk_size`. See also [this vignette](https://cran.r-project.org/web/packages/lidR/vignettes/lidR-computation-speed-LAScatalog.html). The function `catalog_retile` exists but  the doc states:

> Note that this function is not actually very useful since lidR manages everything (clipping, processing, buffering, ...) internally using the proper options. Thus, retiling may be useful for working in other software, for example, but not in lidR.

komazsofi, 2019-01-19:
Yesterday and today I have tested my error occurrence on a smaller dataset. I cannot reproduce the error with the usage of the stdmetrics. The error do NOT occur if I remove the library("e1071") related metrics (zskew = skewness(z),zkurto = kurtosis(z)) from my function. What is also strange if I do not pack together all my metrics functions to compute in one go the error occur. I run the functions one-by-one in separate R files everything is working. So the error specifically occurs during running this particular code:

```r
 library(lidR)

CoverageMetrics = function(z,classification) {
  coveragemetrics = list(
    pulsepenrat = length(z[classification==2])/length(z),
    nofret_abovemean = length(z[z>mean(z)])/length(z)
  )
  return(coveragemetrics)
}

VegStr_VertDistr_Metrics = function(z)
{
  library("e1071")
  vertdistr_metrics = list(
    zstd = sd(z),
    zvar = var(z),
    zskew = skewness(z),
    zkurto = kurtosis(z),
    zentropy2=entropy(z+500, by = 0.5,zmax=NULL)
  )
  return(vertdistr_metrics)
}

HeightMetrics = function(z)
{
  heightmetrics = list(
    zmax = max(z), 
    zmean = mean(z),
    zmedian = median(z),
    z025quantile = quantile(z, 0.25),
    z075quantile = quantile(z, 0.75),
    z090quantile = quantile(z, 0.90)
  )
  return(heightmetrics)
}

ShapeMetrics = function(X,Y,Z)
{
  xyz=rbind(X,Y,Z) 
  cov_m=cov(xyz)
  eigen_m=eigen(cov_m)
  
  shapemetrics = list(
    eigen_largest = eigen_m$values[1],
    eigen_medium = eigen_m$values[2],
    eigen_smallest = eigen_m$values[3],
    curvature = eigen_m$values[3]/(eigen_m$values[1]+eigen_m$values[2]+eigen_m$values[3]),
    linearity = (eigen_m$values[1]-eigen_m$values[2])/eigen_m$values[1],
    planarity = (eigen_m$values[2]-eigen_m$values[3])/eigen_m$values[1],
    sphericity = eigen_m$values[3]/eigen_m$values[1],
    anisotrophy = (eigen_m$values[1]-eigen_m$values[3])/eigen_m$values[1]
  )
  return(shapemetrics)
}

# Set working dirctory
workingdirectory="D:/Koma/Paper1/ALS/lidR_error/"
setwd(workingdirectory)

ctg <- catalog(workingdirectory)

opt_chunk_buffer(ctg) <- 1
opt_chunk_size(ctg) <- 500
opt_cores(ctg) <- 18

metrics = grid_metrics(ctg, c(CoverageMetrics(Z,Classification), VegStr_VertDistr_Metrics(Z), HeightMetrics(Z), ShapeMetrics(X,Y,Z)), res = 2.5)
plot(metrics)
```

I understand that this error is too specific, you do not have time to find out where the problem comes from because of the most possible scenario is that it comes from the user-defined lidar metrics calculation function. I can work around this issue if I need. I think the problem is not the in the feature calculation itself (it is always 100% processed) the error occurs somewhere when the rasters are merged together.

```
Error message:
Processing [================================================================================] 100% (80/80) eta:  0s
Error in dimnames(x) <- dn : 
  length of 'dimnames' [2] not equal to array extent
```
  
If you would like to test this issue I can send you the smaller-set retiled files.
  
Best,
Zsofia

Jean-Romain, 2019-01-19:
You must reduce again the range of possibilities. I ran your code on 30 chunks of one of my dataset (without `ShapeMetrics` which is ways too slow and with `res = 10`) and it worked. Here some tests you can do

1. Find the minimum number of metrics required to reproduce. Remove all others
2. Run only on the first tile, then two tiles, then 3 tiles until it fails. If it fails after 10/80 chunks you don't need to run everything every-time.

```r
ctg1tile = ctg[1,]
ctg2tiles = ctg[1:2,]
# and so on
```
3. Try to use the following. This is a simplified version that do not merge the rasters.

```r
grid_metrics2 = function(las, func, res = 20, start = c(0,0), filter = NULL)
{
  alignment <- list(res = res, start = start)
  lidR::opt_chunk_buffer(las) <- 0.1*alignment$res

  is_formula <- tryCatch(lazyeval::is_formula(func), error = function(e) FALSE)
  if (!is_formula) func <- lazyeval::f_capture(func)
  glob <- future::getGlobalsAndPackages(func)

  options <- list(need_buffer = FALSE, drop_null = TRUE, globals = names(glob$globals), raster_alignment = alignment)
  output  <- lidR::catalog_apply(las, grid_metrics, func = func, res = res, start = start, filter = filter, .options = options)
  return(output)
}
```

4. Merge the output yourself to figure out if the errror occurs during the merge

```r
list_metrics = grid_metrics2(ctg, c(CoverageMetrics(Z,Classification), VegStr_VertDistr_Metrics(Z), HeightMetrics(Z)), res = 10)
r = lidR:::merge_rasters(list_metrics)
```

komazsofi, 2019-01-20:
Hi,

First of all, thank you for the useful and clear suggestions. It was really helpful.

The problem occurs when I am adding the 4th tile. If I run only the 4th tile it is working but if I run all of the 4 tiles together I get the error message reported above. I added one-by-one the LiDAR metrics and the skewness and the kurtosis are the ones where the process is failing. If I am running only the skewness and kurtosis lidar metrics calculation it is working. So all in all this error is really strange and not really consequently reproducible. 

I have also tried your grid_metrics2() function and if I used all of my metrics with all the 4 tiles it is failed in the end with the same error message. That means not the merging is failing. 

I have made a work around the issue, so it does not bother my data processing that much anymore.

Jean-Romain, 2019-01-21:
Great we now have a pretty accurate idea of where does it come from. Could you rerun with tiles 3 and 4 then tile 2,3,4 to figure out the minimum number of tilse you need to share. Then please share the tiles and a minimal reproducible example. I will try to understand where does it come from. For now I don't have any clue. I guess that in some condition `skewness` or `kurtosis` return `NULL` I maybe it fails because of that  :persevere:.

Jean-Romain, 2019-01-22:
It is only 4 Gb of data and area based approach. I guess that I should be able to process that tomorrow on my laptop. Please share the exact minimal code that should enable me to reproduce the bug. I want to be sure that I will not wait too long the end of the computation for nothing.

komazsofi, 2019-01-23:
I've tested whether I can reduce the number of metrics  (again) and I have to say unfortunately you need to run all the metrics calculation function to get to the error.  

I have tried the following scenarios and then everything WORKED WELL (numbers representing the functions e.g. function 1 = : CoverageMetrics() in my code):
- Only 1, Only 2, Only 3, Only 4
- 1+2
-1+2+3

What is NOT WORKING: function 1+2+3+4 using together so you need to run the following code (even it is long to able to see the error (at least on my computer)):

```r
library(lidR)

CoverageMetrics = function(z,classification) {
  coveragemetrics = list(
    pulsepenrat = length(z[classification==2])/length(z),
    nofret_abovemean = length(z[z>mean(z)])/length(z)
  )
  return(coveragemetrics)
}

VegStr_VertDistr_Metrics = function(z)
{
  library("e1071")
  vertdistr_metrics = list(
    zstd = sd(z),
    zvar = var(z),
    zskew = skewness(z),
    zkurto = kurtosis(z),
    zentropy2=entropy(z+500, by = 0.5,zmax=NULL)
  )
  return(vertdistr_metrics)
}

HeightMetrics = function(z)
{
  heightmetrics = list(
    zmax = max(z), 
    zmean = mean(z),
    zmedian = median(z),
    z025quantile = quantile(z, 0.25),
    z075quantile = quantile(z, 0.75),
    z090quantile = quantile(z, 0.90)
  )
  return(heightmetrics)
}

ShapeMetrics = function(X,Y,Z)
{
  xyz=rbind(X,Y,Z) 
  cov_m=cov(xyz)
  eigen_m=eigen(cov_m)
  
  shapemetrics = list(
    eigen_largest = eigen_m$values[1],
    eigen_medium = eigen_m$values[2],
    eigen_smallest = eigen_m$values[3],
    curvature = eigen_m$values[3]/(eigen_m$values[1]+eigen_m$values[2]+eigen_m$values[3]),
    linearity = (eigen_m$values[1]-eigen_m$values[2])/eigen_m$values[1],
    planarity = (eigen_m$values[2]-eigen_m$values[3])/eigen_m$values[1],
    sphericity = eigen_m$values[3]/eigen_m$values[1],
    anisotrophy = (eigen_m$values[1]-eigen_m$values[3])/eigen_m$values[1]
  )
  return(shapemetrics)
}

grid_metrics2 = function(las, func, res = 20, start = c(0,0), filter = NULL)
{
  alignment <- list(res = res, start = start)
  lidR::opt_chunk_buffer(las) <- 0.1*alignment$res
  
  is_formula <- tryCatch(lazyeval::is_formula(func), error = function(e) FALSE)
  if (!is_formula) func <- lazyeval::f_capture(func)
  glob <- future::getGlobalsAndPackages(func)
  
  options <- list(need_buffer = FALSE, drop_null = TRUE, globals = names(glob$globals), raster_alignment = alignment)
  output  <- lidR::catalog_apply(las, grid_metrics, func = func, res = res, start = start,.options = options)
  return(output)
}

# Set working dirctory
#workingdirectory="C:/Koma/Paper1/ALS/"
workingdirectory="D:/Koma/Paper1/ALS/lidR_error/"
setwd(workingdirectory)

ctg <- catalog(workingdirectory)

opt_chunk_buffer(ctg) <- 1
opt_chunk_size(ctg) <- 500
opt_cores(ctg) <- 18

metrics = grid_metrics(ctg,  c(CoverageMetrics(Z,Classification), VegStr_VertDistr_Metrics(Z), HeightMetrics(Z), ShapeMetrics(X,Y,Z)), res = 2.5)
plot(metrics)
```

The error will be:
```
Processing [=======================================================================================================================================================================================================================================] 100% (80/80) eta:  0s
Error in dimnames(x) <- dn : 
  length of 'dimnames' [2] not equal to array extent
```

bi0m3trics, 2019-01-23:
I feel like I ran into this error earlier this month when I was processing a 250k ha landscape to build dem’s and chm’s, nothing fancy. I came to the conclusion that it was the merging of the rasters (made from the clusters) or it had to do with building the virtual raster, but I blew past it since I write the individual results to a file and merge them later and I had other things to do. I don’t have time right now, but I could rerun the code and see if I get the same error… I just know I only ran into this with big landscapes.
-bi0m3trics

Jean-Romain, 2019-01-23:
@komazsofi  `ShapeMetrics` will take forever to compute. And yet, assuming I run the code until the end during the night, this will only allow me to confirm that the error is reproducible but I will need to rerun it several times before to figure out what is going wrong. I have to find a strategy to attack the problem.

@bi0m3trics if you have an example can be reproduced under few minutes of computation this is great.

Jean-Romain, 2019-01-23:
I run you code on your data. Let see what happened tomorrow morning :pray:

komazsofi, 2019-01-23:
I hope the best.

Jean-Romain, 2019-01-24:
I reproduced the bug. Now I must figure out where does it comes from. It is likely to take hours for each trial.

Jean-Romain, 2019-01-24:
:x:   Test 1: metrics 1,2,3,4 + tiles 1,2,3,4
:heavy_check_mark: Test 2: metrics 1,2,3 + tiles 1,2,3,4
:white_large_square: Test 3: metrics 1,2,3,4 + tiles 4
:white_large_square: Test 4: metrics 1,2,3,4 + tiles 3,4
:white_large_square: Test 5: metrics 1,2,3,4 + tiles 2,3,4

closed 2019-01-25 by Jean-Romain

bi0m3trics, 2019-01-25:
@Jean-Romain This might just be your use of english, but I never said it was a workaround (as it necessitated merging the rasters outside of the `raster `package or outside of `R` entirely), but rather that I thought it had to do with merging the rasters and that this might be a place to look. I agree that writing the raster "chunks" to a scratch disk is a good practice, especially if there's a chance something may fail and you don't want to rerun the entire analysis (which is commonly the case in producing first-order products for large landscapes). I've found this functionality (among others) in 2.0.x to be an excellent addition.

Jean-Romain, 2019-01-25:
You did not say that. But in case somebody would interpret your fixes as a workaround I explicitly say that it is not a workaround. It is the good way to process.

bi0m3trics, 2019-01-25:
Gotcha ;)

komazsofi, 2019-01-25:
Thank you for your work and the detailed report about the issue. Now I will follow @bi0m3trics way to process the data.

komazsofi, 2019-01-30:
Seems to be a possible solution, put at the beginning of the code:

rasterOptions(maxmemory = allowed memory in byte)

And with this, the files can be merged.

Jean-Romain, 2019-01-30:
This is indeed another option

Jean-Romain, 2019-01-31:
@komazsofi  would you like to open a new issue to suggest to add a function `shape_metrics` with your metrics. If you open an issue to suggest to add these metrics and you clearly explain me their meaning I could make a much faster implementation natively in lidR. If you are not interested just skip this message.

komazsofi, 2019-01-31:
yes, of course, I will do that

bi0m3trics, 2019-02-01:
@komazsofi I'll second this request... I've seen various publications over the years where eigenvalues (specifically used in curvature analysis - e.g., [Barilotti A, Crosilla F, Sepic F. 2009. Curvature analysis of lidar data for single tree species classification in alpine latitude forests](http://www.isprs.org/proceedings/xxxviii/3-w8/papers/p101.pdf)) has been used to classify tree species and I would imagine that when used under the right coditions it could be shown to imporve prediciton of second-order products (in adddition to traditional z, i, rn lasmetrics and multispectral imagery).

commit 22398eca (2019-01-25): Fix #212