r-lidar/lidR #214: Parallelism with `OpenMP`

Jean-Romain, 2019-01-28:
The option `opt_cores` is badly used and inneficient

1. It loads several chunks at once (memory inefficient)
2. It loads several chunks at once (strong overhead)
3. Is useful only with `LAScatalog`. No multicore when processing `LAS`objects
4. Is badly used by users who believe that more cores = more speed which is often wrong because of 1 & 2.

Thus `opt_cores` will be removed. The catalog processing engine will be rewritten with regular single threaded code  (simpler and safer). Dependency to `future` will be removed.

Instead the C++ code will be parallelized wherever it is possible with `openmp`. `opt_cores` will be replace by `set_lidr_thread()` wich by default will be set to the number of logical core. The overall package will be significantly faster.

I already made tests with `lmf`

``` r
library(lidR)
las = readLAS("~/Documents/ALS data/Montmorency dataset/las/tiles_326500_5227000.las")

# single thread
set_lidr_threads(1)
system.time(tree_detection(las, lmf(4)))
#>  3.143 s

# max cores
set_lidr_threads(0) 
system.time(tree_detection(las, lmf(4))) 
#>  1.737 s
```

Jean-Romain, 2019-01-30:
@bi0m3trics 

```r
LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las <- readLAS(LASfile, select = "xyzi", filter="-keep_first")

# single thread
set_lidr_threads(1)
system.time(lassnags(las, wing2015()))
#>  0.29 s

# max cores
set_lidr_threads(0) 
system.time(lassnags(las, k = 20))
#>  0.14 s

# One of the first R implementation
system.time(lassnags(las, algo = "wing2015"))
#> 2.1 s
```

Jean-Romain, 2019-01-30:
I gained a new factor ~2 on `wing2015` with native C++ parallelization. Maybe more with more thread but I have only 4 cores. That is the result. The old code was only for estimation of the gain from the beginning. Speed-up is like a game. Our score is x15 :smile:

bi0m3trics, 2019-01-30:
That's awesome and very promising - and I look forward 2.1 and studying your code (you seem to never stop teaching me new programming skills).

closed 2019-02-15 by Jean-Romain