r-lidar/lidR #215: problem with custom tree_metrics function

jstrunk001, 2019-01-29:
When I create a custom function for tree_metrics I get the error shown below. I also provide some information about my environment. I updated lidR a few minutes ago, and the error was also present on an earlier version of lidR.

Error in lazyeval::f_capture(func) : Promise has already been forced

RStudio V 1.2.114
R version 3.4.3 (2017-11-30)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 7 x64 (build 7600)
lidR version 2.0.1

R code (taken from the lidR wiki):
```
las1=readLAS("someLas.las")
algo = pitfree(thresholds = c(0,10,20,30,40,50), subcircle = 0.2)
chm  = grid_canopy(las1, 0.5, algo)

# smoothing post-process (e.g. two pass, 3x3 median convolution)
ker = matrix(1,3,3)
chm = focal(chm, w = ker, fun = median)
chm = focal(chm, w = ker, fun = median)

#segment
algo = watershed(chm, th = 6)
x1  = lastrees(las1, algo)

#compute metrics / tree
metric = tree_metrics(x1, .stdtreemetrics)

#create new tree metric function
.treemetrics1 = function (x) 
{
	npoints = length(x)

	metrics = list(npoints = npoints)
	return(metrics)
}

metric1 = tree_metrics(x1, .treemetrics1)
```

Jean-Romain, 2019-01-29:
Hi Jacob,

Sadly I do not answer anymore to questions on github or email. That take me too much time and nobody else can take advantage of my answers. I make a exception for you but the next time please ask on https://gis.stackexchange.com/. Me or somebody else will answer and the answer will be indexed by search engines.

Check the documentation of `grid_metrics` with work exactly the same. This should work

```r
treemetrics1 = function (x) 
{
   npoints = length(x)
   metrics = list(npoints = npoints)
   return(metrics)
}

.treemetrics1 = ~treemetrics1(Z)

metric1 = tree_metrics(x1, .treemetrics1)
```

closed 2019-01-29 by Jean-Romain

jstrunk001, 2019-01-30:
OK! Thank you for the heads up JR! I will direct questions to https://gis.stackexchange.com/ in the future.

Regards,

[Forest Service Shield]

Jacob Strunk, PhD
Statistician

Forest Service
PNW Research Station, Vegetation Monitoring and Remote Sensing Team (VMaRS)

p: 360-218-4285
jacob.strunk@usda.gov<mailto:jacob.strunk@usda.gov>

3625 93rd Ave SW
Olympia, WA 98512
www.fs.fed.us<http://www.fs.fed.us/>
[USDA Logo]<http://usda.gov/> [Forest Service Twitter] <https://twitter.com/forestservice>  [USDA Facebook] <https://www.facebook.com/pages/US-Forest-Service/1431984283714112>

Caring for the land and serving people

This electronic message contains information generated by the USDA solely for the intended recipients. Any unauthorized interception of this message or the use or disclosure of the information it contains may violate the law and subject the violator to civil or criminal penalties. If you believe you have received this message in error, please notify the sender and delete the email immediately.